kuoruan/openwrt-v2ray #2: 编译报错src/v2ray.com/core/config.go:9:2: cannot find package "github.com/golang/protobuf/proto" in any of:

lllrrr, 2019-04-21:
src/v2ray.com/core/config.go:9:2: cannot find package "github.com/golang/protobu                                                      f/proto" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/github.com/golan                                                      g/protobuf/proto (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/github.com/golang/protobuf/proto (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:8:2: cannot find package "go.starlark                                                      .net/starlark" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/                                                      starlark (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/go.starlark.net/starlark (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:9:2: cannot find package "go.starlark                                                      .net/syntax" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/                                                      syntax (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/go.starlark.net/syntax (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/cipher_suites.                                                      go:17:2: cannot find package "golang.org/x/crypto/chacha20poly1305" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/cry                                                      pto/chacha20poly1305 (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/crypto/chacha20poly1305 (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/handshake_mess                                                      ages.go:11:2: cannot find package "golang.org/x/crypto/cryptobyte" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/cry                                                      pto/cryptobyte (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/crypto/cryptobyte (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/key_schedule.g                                                      o:12:2: cannot find package "golang.org/x/crypto/curve25519" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/cry                                                      pto/curve25519 (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/crypto/curve25519 (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/key_schedule.g                                                      o:13:2: cannot find package "golang.org/x/crypto/hkdf" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/cry                                                      pto/hkdf (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/crypto/hkdf (from $GOPATH)
src/v2ray.com/core/proxy/vmess/encoding/auth.go:10:2: cannot find package "golan                                                      g.org/x/crypto/sha3" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/cry                                                      pto/sha3 (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/crypto/sha3 (from $GOPATH)
src/v2ray.com/core/app/dns/udpns.go:12:2: cannot find package "golang.org/x/net/                                                      dns/dnsmessage" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net                                                      /dns/dnsmessage (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/net/dns/dnsmessage (from $GOPATH)
src/v2ray.com/core/transport/internet/http/dialer.go:12:2: cannot find package "                                                      golang.org/x/net/http2" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net                                                      /http2 (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/golang.org/x/net/http2 (from $GOPATH)
src/v2ray.com/core/app/commander/commander.go:12:2: cannot find package "google.                                                      golang.org/grpc" in any of:
        /home/ronlee/182le/staging_dir/hostpkg/lib/go-cross/src/google.golang.or                                                      g/grpc (from $GOROOT)
        /home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.18.0/.                                                      go_work/build/src/google.golang.org/grpc (from $GOPATH)
make[2]: *** [/home/ronlee/182le/build_dir/target-mipsel_24kc_musl/v2ray-core-4.                                                      18.0/.built] Error 1
机器没有被墙，但就是报这个错，不知为什么

kuoruan, 2019-04-21:
需要 master 版 package 源码，或者 https://github.com/openwrt/packages/commit/7dc1f3e0293588ebc544e8eee104043dd0dacaf5 这个 commit 之后的版本

closed 2019-04-21 by kuoruan

linanw, 2019-06-09:
@lllrrr did if fix your issue? By adding packages's source code? Actually it does fix the same issue for me when I build for RPi3. I fixed the issue by adding those go packages manually: 
wget https://dl.google.com/go/go1.12.5.linux-amd64.tar.gz && \
    tar -C /usr/local -xzf go1.12.5.linux-amd64.tar.gz && \
    export PATH=$PATH:/usr/local/go/bin && \
    go get -d github.com/golang/protobuf/proto && \
    go get -d go.starlark.net/starlark && \
    go get -d go.starlark.net/syntax && \
    go get -d golang.org/x/crypto/chacha20poly1305 && \
    go get -d golang.org/x/crypto/cryptobyte && \
    go get -d golang.org/x/crypto/curve25519 && \
    go get -d golang.org/x/crypto/hkdf && \
    go get -d golang.org/x/crypto/sha3 && \
    go get -d golang.org/x/net/dns/dnsmessage && \
    go get -d golang.org/x/net/http2 && \
    go get -d google.golang.org/grpc

cp -r ~/go/src/* ~/openwrt/staging_dir/hostpkg/lib/go-cross/src/

However I still have other issues preventing me from succeed. https://github.com/kuoruan/openwrt-v2ray/issues/9