kuoruan/openwrt-v2ray #5: "pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured"

kokone, 2019-06-02:
你好，

我的中文不太好，所以我使用谷歌翻译下面的原始英文。

首先，谢谢你做到这一点。 真的太棒了！

在尝试从发行版安装时，我收到以下错误：
```
root@OpenWrt:/tmp# curl -L https://github.com/kuoruan/openwrt-v2ray/releases/download/v4.18.2-1/v2ray-core_4.18.2-1_arm_cortex-a9.ipk --output v2ray-core.ipk
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   628    0   628    0     0    539      0 --:--:--  0:00:01 --:--:--   541
100 12.2M  100 12.2M    0     0  20223      0  0:10:36  0:10:35  0:00:01 31833
root@OpenWrt:/tmp# opkg install v2ray-core.ipk
Unknown package 'v2ray-core'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core.
```
`opkg print-architecture`是：
```
arch all 1
arch noarch 1
arch arm_cortex-a9_vfpv3 10
```

我自己尝试从源代码编译，但是我得到了很多我无法弄清楚的错误。

-----

Hello,

My Chinese isn't very good so I'm using Google translate with the original English below.

First, thank you for making this. It's really awesome!

While trying to install from releases, I get the following error:
```
root@OpenWrt:/tmp# curl -L https://github.com/kuoruan/openwrt-v2ray/releases/download/v4.18.2-1/v2ray-core_4.18.2-1_arm_cortex-a9.ipk --output v2ray-core.ipk
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   628    0   628    0     0    539      0 --:--:--  0:00:01 --:--:--   541
100 12.2M  100 12.2M    0     0  20223      0  0:10:36  0:10:35  0:00:01 31833
root@OpenWrt:/tmp# opkg install v2ray-core.ipk
Unknown package 'v2ray-core'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core.
```
`opkg print-architecture` is:
```
arch all 1
arch noarch 1
arch arm_cortex-a9_vfpv3 10
```

I've tried to compile from source myself but I get a lot of errors that I can't figure out.

kuoruan, 2019-06-02:
What's your device model?
Can you paste the content of system information file?
```
cat /usr/lib/os-release
```
or
```
cat /etc/openwrt_release
```

kokone, 2019-06-02:
```
NAME="Ubuntu"
VERSION="16.04.6 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.6 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```
Hope this helps!

kuoruan, 2019-06-02:
I need your router's model and LEDE or Openwrt's info.
Not your PC's system.

kokone, 2019-06-02:
Wow sorry, that was very stupid of me. I was connected to the wrong SSH

```
NAME="OpenWrt"
VERSION="18.06.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 18.06.2"
VERSION_ID="18.06.2"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r7676-cddd7b4c77"
LEDE_BOARD="mvebu/cortexa9"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt 18.06.2 r7676-cddd7b4c77"
```

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.2'
DISTRIB_REVISION='r7676-cddd7b4c77'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='OpenWrt 18.06.2 r7676-cddd7b4c77'
DISTRIB_TAINTS=''
```

kuoruan, 2019-06-02:
Your router belongs to mvebu/cortexa9. All my build is incompatible with it.
I will add a new build.

kokone, 2019-06-02:
You are so kind. Thank you very, very much.

kuoruan, 2019-06-02:
After the build: https://travis-ci.org/kuoruan/openwrt-v2ray/jobs/540398219
You can try download the ipk file with name arm_cortex-a9_vfpv3 from the new release.
Install it and see if it works.

closed 2019-06-02 by kuoruan

kokone, 2019-06-04:
It works, thanks again!

tannx, 2019-10-02:
got same issue, Need Help, 
My Router Model is NETGEAR R6220, 
`
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.4'
DISTRIB_REVISION='r7808-ef686b7292'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 18.06.4 r7808-ef686b7292'
DISTRIB_TAINTS=''

`

CPU's 
``
cpu model               : MIPS 1004Kc V2.15
``

linhanyang, 2019-10-09:
```
root@LEDE:~# cat /usr/lib/os-release
NAME="LEDE"
VERSION="17.01.4, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot 17.01.4"
VERSION_ID="17.01.4"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r3560-79f57e422d"
LEDE_BOARD="brcm2708/bcm2710"
LEDE_ARCH="arm_cortex-a53_neon-vfpv4"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot 17.01.4 r3560-79f57e422d"
```
能否给编译个版本，谢谢。

tannx, 2019-10-09:
> ```
> root@LEDE:~# cat /usr/lib/os-release
> NAME="LEDE"
> VERSION="17.01.4, Reboot"
> ID="lede"
> ID_LIKE="lede openwrt"
> PRETTY_NAME="LEDE Reboot 17.01.4"
> VERSION_ID="17.01.4" [...]

安裝說明上有自編譯的方法， 

' git clone https://github.com/kuoruan/openwrt-v2ray.git package/v2ray-core '

我的路由器儲存空間太小，連Git都裝不下。

commit 60e46d10 (2019-06-02): build(travis): Add build for ARM Cortex-A9

fix #5