kuoruan/openwrt-v2ray #8: 发现几个问题，虚拟内存占用过高，速度上不去，还有MUX关闭后功能不正常

ArcherCX, 2019-06-08:
硬件：WNDR4300v1，Os：OpenWrt 18.06.2

总体物理内存和cpu可以看出负载并不高，但是VSZ占了662m，我怀疑是这个问题导致了连接速度过低，youtube测试连接速度最高直到4800kbps，大约600KB/s，而我单纯用V2ray客户端连接可以达到5-6万kbps，这个差的有点多
[image]

不知道是我的配置问题还是咋整的，一旦关闭mux，有时候访问youtube连不上，有时候连google都连不上，国内直连是没有问题
附上配置：

> {
    "log":
    {
        "access": "/var/log/v2ray_acess.log",
        "error": "/var/log/v2ray_error.log",
        "loglevel": "warning"
    },
    "inbounds": [
    {
        "port": "10000",
        "protocol": "dokodemo-door",
        "settings":
        {
            "network": "tcp,udp",
            "timeout": 30,
            "followRedirect": true
        },
        "sniffing":
        {
            "enabled": true,
            "destOverride": ["http", "tls"]
        }
    }],
    "outbounds": [
    {
        "protocol": "vmess",
        "tag": "ls",
        "mux":
        {
            "enabled": false,
            "concurrency": 8
        },
        "settings":
        {
            "vnext": [
            {
                "address": "www.abc.com",
                "users": [
                {
                    "id": "xxxxxx",
                    "alterId": 64,
                    "level": 0,
                    "security": "auto"
                }],
                "port": 443
            }]
        },
        "streamSettings":
        {
            "wsSettings":
            {
                "path": "/abcd/",
                "headers":
                {
                    "host": "www.abc.com"
                }
            },
            "tlsSettings":
            {
                "allowInsecure": true
            },
            "security": "tls",
            "network": "ws"
        }
    },
    {
        "protocol": "freedom",
        "settings":
        {},
        "tag": "direct"
    },
    {
        "protocol": "blackhole",
        "settings":
        {},
        "tag": "blocked"
    }],
    "routing":
    {
        "domainStrategy": "IPOnDemand",
        "rules": [
        {
            "type": "field",
            "domain": ["geosite:category-ads-all"],
            "outboundTag": "blocked"
        },
        {
            "type": "field",
            "domain": ["geosite:cn"],
            "outboundTag": "direct"
        },
        {
            "type": "field",
            "domain": ["geosite:cn"],
            "outboundTag": "direct"
        },
        {
            "type": "field",
            "protocol": ["bittorrent"],
            "outboundTag": "direct"
        },
        {
            "type": "field",
            "domain": ["domain:cloudflare.com"],
            "outboundTag": "direct"
        }]
    }
}

kuoruan, 2019-06-09:
你先用官方的编译版本试一试，如果也这样的话，在官方开个 Issue 问问。
如果没这个问题的话，我再看看是不是编译的问题

ArcherCX, 2019-06-09:
官方的版本运行提示illegal instruction，好像是Openwrt需要支持MIPS FPU Emulator，我先看看v2ray的编译，编译一个支持MIPS FPU Emulator的试试吧

ArcherCX, 2019-06-09:
心累，今天重新编译了开启MIPS FPU Emulator的固件，尝试了官方编译的版本，重新编了v2ray-core，一样，最后发现网速问题应该是cpu性能不行了。。。跑到600kb/s差不多就占用90%了，这差距太绝望了，回头买个新路由吧。。。
Mux的问题还是一样，一旦关闭，就存在连接不通的问题，这个我去官方的git上开个issue再问问吧

closed 2019-06-09 by ArcherCX

andrew0x0, 2020-01-16:
wndr3700v4 这货确实没fpu是个硬伤,去掉mux后,我拿着凑合看网页还行,看视频卡的不像话.

kukudemajia, 2020-04-04:
我也遇到类似问题，用koolshare就正常，我是虚拟机跑的，性能不是问题

kuoruan, 2020-04-04:
姑且加了个内存限制的选项，不知道是否有用