kuoruan/openwrt-v2ray #9: Try to build for aarch64_cortex-a53 (RPi 3), but always fails.

linanw, 2019-06-09:
Having this error, not sure if any one had even successfully built for aarch64_cortex-a53 (RPi 3)?

.....
v2ray.com/core/transport/internet/headers/http
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/ackhandler
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/flowcontrol
v2ray.com/core/external/github.com/lucas-clemente/quic-go/internal/handshake
v2ray.com/core/external/github.com/gorilla/websocket
v2ray.com/core/external/github.com/lucas-clemente/quic-go
google.golang.org/grpc/internal/transport
google.golang.org/grpc
make[2]: *** [Makefile:290: /root/openwrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.19.1/.built] Error 2
make[2]: Leaving directory '/root/openwrt/package/v2ray'
time: package/v2ray/compile#26.38#30.08#45.26
make[1]: *** [package/Makefile:109: package/v2ray/compile] Error 2
make[1]: Leaving directory '/root/openwrt'
make: *** [/root/openwrt/include/toplevel.mk:211: package/v2ray/compile] Error 2
root@2a880a08a0da:~/openwrt#

kuoruan, 2019-06-10:
Did you try set proxy when build?

linanw, 2019-06-10:
> Did you try set proxy when build?

Hi Kuoruan, thanks for the super fast response. Nope, I don't have any proxy setting, actually I didn't get what proxy do you mean? I'm with direct network to ISP in Singapore. I built the code with a docker container I built myself with Ubuntu.

kuoruan, 2019-06-10:
From openwrt hwdata, Raspberry Pi 3 use target **brcm2708**, arm_cortex-a53_neon-vfpv4
https://openwrt.org/toh/hwdata/raspberry_pi_foundation/raspberry_pi_3_bplus
Why you build for aarch64_cortex-a53?

linanw, 2019-06-10:
This used the SDK download from https://downloads.openwrt.org/releases/18.06.2/targets/brcm2708/bcm2710/.

aarch64_cortex-a53 is the only toolchain from it’s staging_dir/

Btw: I checked the 18.06 packages for arm_cortex-a53_neon-vfpv4, it is actually empty.

https://downloads.openwrt.org/releases/packages-18.06/arm_cortex-a53_neon-vfpv4/ (empty link)

Regards,
Bruce

https://openwrt.org/toh/hwdata/raspberry_pi_foundation/raspberry_pi_3_bplus<https://eur04.safelinks.protection.outlook.com/?url=https%3A%2F%2Fopenwrt.org%2Ftoh%2Fhwdata%2Fraspberry_pi_foundation%2Fraspberry_pi_3_bplus&data=02%7C01%7C%7Cc61d05a97967456431b908d6ed531493%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636957338411314207&sdata=nb92M0bVPsR7I8amsAn2BLl%2B%2FObzDpFPVbuWj38JIeQ%3D&reserved=0>

kuoruan, 2019-06-10:
Build test passed with snapshot sdk in
https://downloads.openwrt.org/snapshots/targets/brcm2708/bcm2710/

kuoruan, 2019-06-10:
Golang modules support added on openwrt/packages@7dc1f3e0293588ebc544e8eee104043dd0dacaf5
So you need a sdk after this date. Or update the golang package to latest
https://github.com/openwrt/packages/tree/master/lang/golang

commit 5faf8e17 (2019-06-10): chore: Add sdk link and requires commit

close #9
closed 2019-06-10 by kuoruan