kuoruan/openwrt-v2ray #11: 请教大神！关于斐讯N1盒子安装a53架构ipk的问题

jerryjdp, 2019-06-24:
斐讯N1盒子刷了armbian，然后docker了https://downloads.openwrt.org/releases/18.06.2/targets/armvirt/64/openwrt-18.06.2-armvirt-64-default-rootfs.tar.gz
作为旁路由，但是在安装您的a53架构的v2ray.ipk时，总是出错，显示架构不对，多次修改也搞不定，请教大神。

顺便说明一下，N1的cpu是晶晨的S905D，看了说明书显示应该属于a53 neon-vfv4类，可能和您之前回答那个问题的rpi 3差不多架构，但就是装不上去。

谢谢大神！

jerryjdp, 2019-06-24:
补充一下，本来我以为是缺少依赖或者opkg.conf有问题，参照下边这个文章进行了修改，还是不行
https://github.com/dafeiyoung/sguclient/wiki/%E4%BF%AE%E6%94%B9opkg.conf%E8%A7%A3%E5%86%B3incompatible-with-the-architectures-configured%E9%97%AE%E9%A2%98

kuoruan, 2019-06-24:
```
opkg print-architecture
```
```
cat /usr/lib/os-release /etc/openwrt_release
```

jerryjdp, 2019-06-25:
> ```
> opkg print-architecture
> ```
> 
> ```
> cat /usr/lib/os-release /etc/openwrt_release
> ```

大神，非常感谢，今晚回去试一下！

jerryjdp, 2019-06-25:
感谢大神，已经装上了，不过本人小白，还有两个问题需要请教您（N1 armbian dokcer openwrt）

1.装上后运行就出现如下信息
root@OpenWrt:~# v2ray
V2Ray 4.19.1 (Po) OpenWrt - Release 2
A unified platform for anti-censorship.
main: failed to load config:  > v2ray.com/core/main/confloader/external: config file not readable > open : no such file or directory
请问这个配置文件是不是config.json文件？谢谢！

2.请问是否有类似luci-app-v2ray_1.0-5_all.ipk之类的文件，能够在openwrt中方便的管理。

再次感谢大神指点！

jerryjdp, 2019-06-28:
不好意思，耽误您宝贵时间了！

还是想请教一下，目前已经安装上了，也加入了配置文件，但运行起来还是无法作为旁路由来用，不知道什么原因

kuoruan, 2019-06-28:
具体的配置问题，自行百度
旁路由的话，需要做成透明代理

jerryjdp, 2019-06-28:
> 具体的配置问题，自行百度
> 旁路由的话，需要做成透明代理

好的，谢谢！我先自己学习一下！

closed 2019-06-29 by jerryjdp