kuoruan/openwrt-v2ray #12: 大神，装是装上了，可否做成安装时直接装成服务或者开机自动执行。

sparkyuanquan, 2019-06-28:
你好，大神
查了好多资料，只有您的APK布属在软路由上可以运行。感谢您的努力，小白用户，可能请教的问题比较初级，还请见谅。
我是想在软路由 X86的机器上Openwrt Lede上装V2ray 做为服务端，放在HK的朋友家。这样可以避免地址被墙的问题，HK的宽带是自动获取动态公网地址，就算被墙也只要重起一下就可以了。试了V2ray官网的脚本wget https://install.direct/go.sh，在Lede上 ，装是可以装上，但是无法起动V2ray,提示权限拒绝。
然后改用了您的APK文件安装，很顺利的安装到了 /usr/bin 目录下，放入了Config.json 文件后，手动打 V2ray ,可以运行，但是问题是终端上就无法做其它操作了，除非Ctrl+C， 
麻烦问一下,
1  是否可以做成安装时，直接添加成服务，开机自动起动，
2  起动与停止的指令，可以麻烦告知一下吗？

[image]

kuoruan, 2019-06-28:
related to #6

kuoruan, 2019-06-28:
启动的时候在运行命令后面加上 & 可以转入后台运行。

sparkyuanquan, 2019-06-28:
> 启动的时候在运行命令后面加上 & 可以转入后台运行。

恩，好的，加了& 可以后台了：）非常感谢。 继续学习。

lizhongit, 2019-07-04:
如何在酷软中心安装上？这样的话通过酷软中心的界面操作方便

kuoruan, 2019-07-04:
LuCI 初版开发完毕，欢迎测试：https://github.com/kuoruan/luci-app-v2ray/releases/tag/v1.0.0-1

sparkyuanquan, 2019-07-05:
> LuCI 初版开发完毕，欢迎测试：https://github.com/kuoruan/luci-app-v2ray/releases/tag/v1.0.0-1

在虚拟机上测了一下，可以装上。
[image]

sparkyuanquan, 2019-07-05:
感谢大神的辛苦劳动：）

rny, 2019-07-05:
```
root@OpenWrt:/tmp# uname -r
4.9.184
root@OpenWrt:/tmp# opkg install luci-app-v2ray.ipk
Installing luci-app-v2ray (1.1.0-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-v2ray:
 *      kernel (= 4.9.184-1-2dda2a2d0ee2eba71176fe2e0f6d5dff)
 * opkg_install_cmd: Cannot install package luci-app-v2ray.
```

fix by using 

```
root@OpenWrt:/tmp# opkg install luci-app-v2ray.ipk --force-depends
```

closed 2019-07-06 by kuoruan