kuoruan/openwrt-v2ray #14: 開啟服務后，路由器上原本的端口映射失效。

DaimonBaby, 2019-08-03:
我的是小米路由器，是外网ip，而且已經配置好ddns，設置好路由器的端口映射，啟動服務前，外網能鏈接上內網的設備，開啟服務后就鏈接不上了，請問大大是不是軟件裡面的路由功能也要設置？還是要命令行裡面手動設置iptable?

DaimonBaby, 2019-08-03:
現在是使用您的luci，沒有什麼問題，只是斷連厲害，家裡熱水器一直在叫而已。現在最大的問題就是端口映射問題了。

DaimonBaby, 2019-08-03:
断连可能是我的路由器性能差，链接数一上来就不行了，但是每次热水器断连后都会马上连上了，在本软件的路由功能上已经把热水器踢出去了，但是还是在叫，还有端口映射的设备我都已经踢了出去的
{
				"type": "field",
				"port": "5000-60000",
				"outboundTag": "direct"
			},
			{
				"type": "field",
				"source": [
					"192.168.1.231",
					"192.168.1.147",
					"192.168.1.225"
				],
				"outboundTag": "direct"
			}

kuoruan, 2019-08-04:
是不是在 luci 上开启了透明代理？开启之后，本地所有的数据都会经过 v2ray，而你的路由器性能比较若，会经常断连。
可以试试通过 iptables 直接忽略这些连接。
```
iptables -A V2RAY -s 192.168.1.231 -j RETURN
```

DaimonBaby, 2019-08-04:
好的，谢谢。是开了透明代理，今晚回家试试设置iptable

DaimonBaby, 2019-08-04:
端口映射问题也是忽略链接就可以解决么？我今晚回去试试忽略了之后看看外网能不能连上内网。

DaimonBaby, 2019-08-04:
root@SioHome:~# iptables -t V2RAY -L -n
iptables v1.6.2: can't initialize iptables table `V2RAY': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@SioHome:~# iptables -t v2ray -L -n
iptables v1.6.2: can't initialize iptables table `v2ray': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
root@SioHome:~# iptables -A V2RAY -s 192.168.1.231 -j RETURN
iptables: No chain/target/match by that name.
root@SioHome:~# iptables -A v2ray -s 192.168.1.231 -j RETURN
iptables: No chain/target/match by that name.
不行，没有这个表

DaimonBaby, 2019-08-04:
可以了，添加上表名就可以了。
`iptables -t nat -A V2RAY -s 192.168.1.225 -j RETURN`

closed 2019-08-04 by DaimonBaby