kuoruan/openwrt-v2ray #15: can NOT build on latest official openwrt release v18.06.4

moralrebuild, 2019-08-06:
Please help correct the feeds or patch the source codes to support the latest official openwrt release. Otherwise this app fails to compile with following golang issue:

```
src/v2ray.com/core/config.go:9:2: cannot find package "github.com/golang/protobuf/proto" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/github.com/golang/protobuf/proto (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/github.com/golang/protobuf/proto (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:8:2: cannot find package "go.starlark.net/starlark" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/starlark (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/go.starlark.net/starlark (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:9:2: cannot find package "go.starlark.net/syntax" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/syntax (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/go.starlark.net/syntax (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/cipher_suites.go:17:2: cannot find package "golang.org/x/crypto/chacha20poly1305" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/chacha20poly1305 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/chacha20poly1305 (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/handshake_messages.go:11:2: cannot find package "golang.org/x/crypto/cryptobyte" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/cryptobyte (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/cryptobyte (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/key_schedule.go:12:2: cannot find package "golang.org/x/crypto/curve25519" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/curve25519 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/curve25519 (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/key_schedule.go:13:2: cannot find package "golang.org/x/crypto/hkdf" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/hkdf (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/hkdf (from $GOPATH)
src/v2ray.com/core/proxy/vmess/encoding/auth.go:10:2: cannot find package "golang.org/x/crypto/sha3" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/sha3 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/sha3 (from $GOPATH)
src/v2ray.com/core/app/dns/udpns.go:12:2: cannot find package "golang.org/x/net/dns/dnsmessage" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/dns/dnsmessage (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/net/dns/dnsmessage (from $GOPATH)
src/v2ray.com/core/transport/internet/http/dialer.go:12:2: cannot find package "golang.org/x/net/http2" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/http2 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/net/http2 (from $GOPATH)
src/v2ray.com/core/transport/internet/http/hub.go:21:2: cannot find package "golang.org/x/net/http2/h2c" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/http2/h2c (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/net/http2/h2c (from $GOPATH)
src/v2ray.com/core/app/commander/commander.go:12:2: cannot find package "google.golang.org/grpc" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/google.golang.org/grpc (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/google.golang.org/grpc (from $GOPATH)
```

kuoruan, 2019-08-06:
Need commit openwrt/packages@7dc1f3e0293588ebc544e8eee104043dd0dacaf5
You can just update your golang package to latest: https://github.com/openwrt/packages/tree/master/lang/golang

closed 2019-08-06 by kuoruan

moralrebuild, 2019-08-06:
@kuoruan Thanks for your quick support. Could you please elaborate how to do that?

kuoruan, 2019-08-06:
```
wget -O feeds/packages/lang/golang/golang-package.mk https://github.com/openwrt/packages/raw/master/lang/golang/golang-package.mk
```

moralrebuild, 2019-08-06:
Thank @kuoruan!
But it looks I need to do something more than just executing your command and resuming the compiling:

```

Building targets
src/v2ray.com/core/config.go:9:2: cannot find package "github.com/golang/protobuf/proto" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/github.com/golang/protobuf/proto (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/github.com/golang/protobuf/proto (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:8:2: cannot find package "go.starlark.net/starlark" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/starlark (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/go.starlark.net/starlark (from $GOPATH)
src/v2ray.com/core/app/router/condition.go:9:2: cannot find package "go.starlark.net/syntax" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/go.starlark.net/syntax (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/go.starlark.net/syntax (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/cipher_suites.go:17:2: cannot find package "golang.org/x/crypto/chacha20poly1305" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/chacha20poly1305 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/chacha20poly1305 (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/handshake_messages.go:11:2: cannot find package "golang.org/x/crypto/cryptobyte" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/cryptobyte (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/cryptobyte (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/key_schedule.go:12:2: cannot find package "golang.org/x/crypto/curve25519" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/curve25519 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/curve25519 (from $GOPATH)
src/v2ray.com/core/external/github.com/refraction-networking/utls/key_schedule.go:13:2: cannot find package "golang.org/x/crypto/hkdf" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/hkdf (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/hkdf (from $GOPATH)
src/v2ray.com/core/proxy/vmess/encoding/auth.go:10:2: cannot find package "golang.org/x/crypto/sha3" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/crypto/sha3 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/crypto/sha3 (from $GOPATH)
src/v2ray.com/core/app/dns/udpns.go:12:2: cannot find package "golang.org/x/net/dns/dnsmessage" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/dns/dnsmessage (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/net/dns/dnsmessage (from $GOPATH)
src/v2ray.com/core/transport/internet/http/dialer.go:12:2: cannot find package "golang.org/x/net/http2" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/http2 (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/net/http2 (from $GOPATH)
src/v2ray.com/core/transport/internet/http/hub.go:21:2: cannot find package "golang.org/x/net/http2/h2c" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/golang.org/x/net/http2/h2c (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/golang.org/x/net/http2/h2c (from $GOPATH)
src/v2ray.com/core/app/commander/commander.go:12:2: cannot find package "google.golang.org/grpc" in any of:
        /home/wings/openwrt2019/staging_dir/hostpkg/lib/go-cross/src/google.golang.org/grpc (from $GOROOT)
        /home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.go_work/build/src/google.golang.org/grpc (from $GOPATH)

Cleaning module download cache (golang/go#27455)
flag provided but not defined: -modcache
usage: clean [-i] [-r] [-n] [-x] [-cache] [-testcache] [build flags] [packages]
Run 'go help clean' for details.

Makefile:295: recipe for target '/home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.built' failed
make[3]: *** [/home/wings/openwrt2019/build_dir/target-mipsel_24kc_musl/v2ray-core-4.20.0/.built] Error 1
make[3]: Leaving directory '/home/wings/openwrt2019/package/v2ray-core'
```

kuoruan, 2019-08-06:
Replace the golang package in ```feeds/packages/lang/golang```
with the latest one.

kuoruan, 2019-08-06:
Or maybe try:
```
wget -O feeds/packages/lang/golang/golang-package.mk https://github.com/openwrt/packages/raw/7dc1f3e0293588ebc544e8eee104043dd0dacaf5/lang/golang/golang-package.mk
``` 

Then:
```
make package/v2ray-core/{clean,compile} V=s
```

moralrebuild, 2019-08-06:
> 
> 
> Replace the golang package in `feeds/packages/lang/golang`
> with the latest one.

May I have the specific instructions? 

> 
> 
> Or maybe try:
> 
> ```
> wget -O feeds/packages/lang/golang/golang-package.mk https://github.com/openwrt/packages/raw/7dc1f3e0293588ebc544e8eee104043dd0dacaf5/lang/golang/golang-package.mk
> ```
> 
> Then:
> 
> ```
> make package/v2ray-core/{clean,compile} V=s
> ```

No joy

moralrebuild, 2019-08-06:
> 
> 
> Replace the golang package in `feeds/packages/lang/golang`
> with the latest one.

May I have the specific instructions? Thanks.

kuoruan, 2019-08-06:
```
rm -rf feeds/package/lang/golang
git clone -b master --depth 1 https://github.com/openwrt/packages.git ~/openwrt-packages
cp -rf ~/openwrt-packages/lang/golang feeds/package/lang/golang
rm -rf tmp
./script/feeds/install -a
make package/v2ray-core/{clean,compile} V=s
```

moralrebuild, 2019-08-06:
Thanks @kuoruan for your patience. Does you commands mean to build standalone binary on top of the openwrt master branch rather than have it built into the openwrt 18.06.04 single image file?

kuoruan, 2019-08-06:
These commands to replace your golang package with the master branch one.

If you do not want to do that,
use the snapshot sdk to build your image:
https://downloads.openwrt.org/snapshots/

moralrebuild, 2019-08-06:
Thanks a lot for your time and kind support! It seems both go and v2ray are far from mature for average openwrt users.

moralrebuild, 2019-08-07:
@kuoruan 

I tried your last commands again. But found missing something like ''golang/host''
```
./scripts/feeds install -a
ERROR: please fix package/v2ray-core/Makefile - see logs/package/v2ray-core/dump.txt for details
Collecting package info: doneing...eds/telephony/yatedspytmpxtable-mod-bcg729
Collecting target info: doneing...ux/x86xishioe
WARNING: Makefile 'package/v2ray-core/Makefile' has a build dependency on 'golang/host', which does not exist
Installing all packages from feed packages.
Installing package 'golang' from packages
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
``