kuoruan/openwrt-v2ray #17: 树莓派4B安装A72报错

FattyboyN, 2019-08-10:
折腾了两天树莓派4B，终于安装上了OpenWRT。使用的是
http://downloads.openwrt.org/snapshots/targets/brcm2708/bcm2709/openwrt-brcm2708-bcm2709-rpi-2-ext4-factory.img.gz

openWRT官网（https://openwrt.org/toh/hwdata/raspberry_pi_foundation/raspberry_pi_foundation_raspberry_pi_4_b）给的是用http://downloads.openwrt.org/snapshots/targets/brcm2708/bcm2709/openwrt-brcm2708-bcm2709-rpi-2-squashfs-factory.img.gz来安装，但是这样莫名出现SD卡只读错误。换成ext4的就好了。

然后新的问题是版载BCM43455 wifi芯片无法启动。折腾了很久，发现必须将Channel选一个固定的，不能选Auto，否则就无法启动。

最后，安装为4B编译的v2ray ipk （A72），报错：
*pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core.

切换了其他几个，都是同样的。

检查opkg update，发现走的是
http://downloads.openwrt.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/Packages.gz

所以这个snapshot是否需要用到arm-cortex-a7_neon才能安装？

麻烦了

kuoruan, 2019-08-10:
好吧，我再加上 arm-cortex-a7_neon 的编译

FattyboyN, 2019-08-10:
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch arm_cortex-a7_neon-vfpv4 10

FattyboyN, 2019-08-11:
> 好吧，我再加上 arm-cortex-a7_neon 的编译

谢谢大神，辛苦了！

closed 2019-09-06 by kuoruan