kuoruan/openwrt-v2ray #19: 请问如何选择目标架构呢？我需要编译的架构为 mipsel_1004kc，MT7621处理器用的

lpflhh, 2019-08-17:
请问如何选择目标架构呢？我需要编译的架构为 mipsel_1004kc，MT7621处理器用的，我在编译菜单里没找到，可否告知一下？谢谢了

kuoruan, 2019-08-18:
mipsel_1004kc 是 OpenWrt 15.05 及以下版本的
用这个 SDK
https://archive.openwrt.org/chaos_calmer/15.05/ramips/mt7621/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64.tar.bz2

编译 v2ray 需要最新的 golang 包

closed 2019-08-18 by kuoruan

lpflhh, 2019-08-30:
头都要炸了，从配置golang，到执行编译，各种错误层出不穷，这10天对我这菜鸟来说太难熬了，作者可以帮我编译一下吗？如果可以的话，在下万分感谢！

abc@tulip-pc:/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64$ make package/v2ray-core/{clean,compile} V=s
tmp/.config-package.in:14441:warning: ignoring type redefinition of 'PACKAGE_luci' from 'boolean' to 'tristate'
tmp/.config-package.in:14494:warning: ignoring type redefinition of 'PACKAGE_luci-base' from 'boolean' to 'tristate'
tmp/.config-package.in:14596:warning: ignoring type redefinition of 'PACKAGE_luci-mod-admin-full' from 'boolean' to 'tristate'
tmp/.config-package.in:14937:warning: ignoring type redefinition of 'PACKAGE_luci-app-firewall' from 'boolean' to 'tristate'
tmp/.config-package.in:15637:warning: ignoring type redefinition of 'PACKAGE_luci-theme-bootstrap' from 'boolean' to 'tristate'
tmp/.config-package.in:15700:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ipv6' from 'boolean' to 'tristate'
tmp/.config-package.in:15725:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ppp' from 'boolean' to 'tristate'
tmp/.config-package.in:15766:warning: ignoring type redefinition of 'PACKAGE_luci-lib-ip' from 'boolean' to 'tristate'
tmp/.config-package.in:15819:warning: ignoring type redefinition of 'PACKAGE_luci-lib-nixio' from 'boolean' to 'tristate'
#
# configuration written to .config
#
make[1]: Entering directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
make[2]: Entering directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/v2ray-core'
rm -f /media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ramips/packages/base/v2ray-core_*
rm -f /media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/bin/ramips/packages/base/golang-v2ray-core-dev_*
rm -f /media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/target-mipsel_1004kc+dsp_uClibc-0.9.33.2/stamp/.v2ray-core_installed
rm -f /media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/target-mipsel_1004kc+dsp_uClibc-0.9.33.2/packages/v2ray-core.list /media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/staging_dir/host/packages/v2ray-core.list
rm -rf /media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/build_dir/target-mipsel_1004kc+dsp_uClibc-0.9.33.2/v2ray-core-4.20.0
make[2]: Leaving directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/package/v2ray-core'
make[1]: Leaving directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
tmp/.config-package.in:14441:warning: ignoring type redefinition of 'PACKAGE_luci' from 'boolean' to 'tristate'
tmp/.config-package.in:14494:warning: ignoring type redefinition of 'PACKAGE_luci-base' from 'boolean' to 'tristate'
tmp/.config-package.in:14596:warning: ignoring type redefinition of 'PACKAGE_luci-mod-admin-full' from 'boolean' to 'tristate'
tmp/.config-package.in:14937:warning: ignoring type redefinition of 'PACKAGE_luci-app-firewall' from 'boolean' to 'tristate'
tmp/.config-package.in:15637:warning: ignoring type redefinition of 'PACKAGE_luci-theme-bootstrap' from 'boolean' to 'tristate'
tmp/.config-package.in:15700:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ipv6' from 'boolean' to 'tristate'
tmp/.config-package.in:15725:warning: ignoring type redefinition of 'PACKAGE_luci-proto-ppp' from 'boolean' to 'tristate'
tmp/.config-package.in:15766:warning: ignoring type redefinition of 'PACKAGE_luci-lib-ip' from 'boolean' to 'tristate'
tmp/.config-package.in:15819:warning: ignoring type redefinition of 'PACKAGE_luci-lib-nixio' from 'boolean' to 'tristate'
#
# configuration written to .config
#
make[1]: Entering directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
make[2]: Entering directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/feeds/packages/lang/golang/golang'
Makefile:271: *** go-bootstrap cannot be installed on /amd64.  Stop.
make[2]: Leaving directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/feeds/packages/lang/golang/golang'
package/Makefile:191: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[1]: *** [package/feeds/packages/golang/host/compile] Error 2
make[1]: Leaving directory '/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64'
/media/abc/develop/OpenWrt-SDK-15.05-ramips-mt7621_gcc-4.8-linaro_uClibc-0.9.33.2.Linux-x86_64/include/toplevel.mk:174: recipe for target 'package/v2ray-core/compile' failed
make: *** [package/v2ray-core/compile] Error 2

kuoruan, 2019-08-31:
你这个架构的想用的话，直接下官方的文件即可，没必要自己编译。

lpflhh, 2019-08-31:
我想配合大神的 kuoruan/luci-app-v2ray 控制面板来运行v2ray，官方的虽然可以运行但是和 kuoruan/luci-app-v2ray 面板配合不太好，所以我才想自己编译曲线救国的，话说用官方v2ray配合 kuoruan/luci-app-v2ray 不能在控制界面里启动的。

kuoruan, 2019-09-01:
可以的，没有这个问题

lpflhh, 2019-09-01:
谢谢提醒，我再研究一下

Talaxy009, 2019-12-05:
请问哪里能找到适合MT7620的ipk吗？我在官方网站只能找到一堆压缩包，不会用欸。。。望能指点迷津！