kuoruan/openwrt-v2ray #20: 自己编译总是失败，下载不到"dlc.dat"

ChrisLv-CN, 2019-08-24:
我用的是WNDR3700v4，查到的架构是mips74kc，用大佬编好的那些包都提示架构不对，装不了。固件版本是OpenWrt Chaos Calmer 15.05.1-gy408-V0.0.3 r49599

去下载了ar71xx的sdk。按照说明编译了半天。
最后还是失败了。
`
2019-08-24 17:03:58 (9.42 MB/s) - '/usr/local/openwrt/openwrt-sdk-ar71xx-generic_gcc-7.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/v2ray-core-4.20.0/release/config/geoip.dat' saved [5182377/5182377]

wget -O /usr/local/openwrt/openwrt-sdk-ar71xx-generic_gcc-7.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/v2ray-core-4.20.0/release/config/geosite.dat https://github.com/v2ray/domain-list-community/releases/latest/download/dlc.dat
--2019-08-24 17:03:58--  https://github.com/v2ray/domain-list-community/releases/latest/download/dlc.dat
Resolving github.com (github.com)... 192.30.255.112
Connecting to github.com (github.com)|192.30.255.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github.com/v2ray/domain-list-community/releases/download/20190822.1/dlc.dat [following]
--2019-08-24 17:03:58--  https://github.com/v2ray/domain-list-community/releases/download/20190822.1/dlc.dat
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 404 Not Found
2019-08-24 17:03:58 ERROR 404: Not Found.

make[2]: *** [/usr/local/openwrt/openwrt-sdk-ar71xx-generic_gcc-7.4.0_musl.Linux-x86_64/build_dir/target-mips_24kc_musl/v2ray-core-4.20.0/.prepared_a2bcf72f98f4df82a1e76fd00cea0f4d_18f1e190c5d53547fed41a3eaa76e9e9] Error 8
make[2]: Leaving directory `/usr/local/openwrt/openwrt-sdk-ar71xx-generic_gcc-7.4.0_musl.Linux-x86_64/package/v2ray-core'
time: package/v2ray-core/compile#0.83#0.61#4.69
make[1]: *** [package/v2ray-core/compile] Error 2
make[1]: Leaving directory `/usr/local/openwrt/openwrt-sdk-ar71xx-generic_gcc-7.4.0_musl.Linux-x86_64'
make: *** [package/v2ray-core/compile] Error 2
`

ChrisLv-CN, 2019-08-24:
自己弄好了，原来是最新编译的0822的包里没有这个文件，手工指定为前一个0815的就好了...

closed 2019-08-24 by ChrisLv-CN

kuoruan, 2019-08-25:
这是那边 release 的时候没有传这个文件、、我看看能不能问问怎么回事