kuoruan/openwrt-v2ray #21: V2RAY can not start after router reboot

moralrebuild, 2019-09-04:
Manual start works.  The V2RAY has already been enabled in Global Settings.  The openwrt version is 18.06.4.

kuoruan, 2019-09-04:
Is this a issue for luci-app-v2ray?

Already enabled when install luci

https://github.com/kuoruan/luci-app-v2ray/blob/master/Makefile#L35

moralrebuild, 2019-09-04:
Yes, it's LuCI related.

moralrebuild, 2019-09-04:
B.T.W. I saw two $ when test the variable, is it a typo?

```
if [ -z "**$$**{IPKG_INSTROOT}" ]; then
	( . /etc/uci-defaults/40_luci-v2ray ) && rm -f /etc/uci-defaults/40_luci-v2ray
	chmod 755 /etc/init.d/v2ray >/dev/null 2>&1
	/etc/init.d/v2ray enable >/dev/null 2>&1
fi
```

kuoruan, 2019-09-04:
It's not a typo. It is used to define a shell variable in Makefile

moralrebuild, 2019-09-04:
OK. I had ticked the 'Enabled' switch back and forth, not sure if the code can handle such common operations properly.

closed 2019-09-05 by kuoruan