kuoruan/openwrt-v2ray #23: 获取不了版本信息咋回事

hubingnan, 2019-09-17:
无法获取 V2Ray 版本信息

kuoruan, 2019-09-17:
需要具体描述

或者 在控制台运行：
```
v2ray --version
```

hubingnan, 2019-09-17:
> 需要具体描述
> 
> 或者 在控制台运行：
> 
> ```
> v2ray --version
> ```

https://github.com/kuoruan/openwrt-v2ray/issues/23#issuecomment-532136143

hubingnan, 2019-09-17:
[image]

kuoruan, 2019-09-17:
什么架构？

在命令行运行一下

```
v2ray --version
```

看看输出结果

hubingnan, 2019-09-17:
[image]
[image]

kuoruan, 2019-09-17:
貌似 k2p 和 k3 都不支持，用官方的 arm 编译版本即可

closed 2019-09-17 by kuoruan