kuoruan/openwrt-v2ray #28: wndr4300应该安装哪个版本啊？

user-evan-chen, 2019-09-29:
相当困惑，不知道该安装哪个版本。请指点。谢谢

kuoruan, 2019-10-07:
貌似是 ar71xx，但我没有提供对应的 ipk，安装官方版本试试

closed 2019-10-07 by kuoruan

kastnerka, 2020-06-07:
added arch mips_24kc 100 to /etc/opkg.conf

-----
Installing luci-app-v2ray (1.5.6) to root...
Installing resolveip (2) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/mips_24kc/base/                                          resolveip_2_mips_24kc.ipk
Collected errors:
 * opkg_download: Failed to download https://openwrt.proxy.ustclug.org/snapshots                                          /packages/mips_24kc/base/resolveip_2_mips_24kc.ipk, wget returned 4.
 * opkg_download: Check your network settings and connectivity.