kuoruan/openwrt-v2ray #31: opkg_install_cmd: Cannot install package v2ray-core.

linhanyang, 2019-10-09:
```
root@LEDE:~# cat /usr/lib/os-release
NAME="LEDE"
VERSION="17.01.4, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot 17.01.4"
VERSION_ID="17.01.4"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r3560-79f57e422d"
LEDE_BOARD="brcm2708/bcm2710"
LEDE_ARCH="arm_cortex-a53_neon-vfpv4"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot 17.01.4 r3560-79f57e422d"
```
能否给编译个版本，谢谢。

kuoruan, 2019-10-09:
先用着官方版吧，我后期在加上

kuoruan, 2019-10-12:
查了一下官方文档，你的路由器需要安装 aarch64_cortex-a53
https://openwrt.org/docs/techref/instructionset/aarch64_cortex-a53

应该是新版改了。

closed 2019-10-12 by kuoruan