kuoruan/openwrt-v2ray #33: Illegal Instruction issue

steavenku, 2019-10-12:
I was using Netgear R6300 V2 with OpenWrt 17.01.4. 
I can install the v2ray package, but when I run from /usr/bin/v2ray, it shows the illegal instruction. Is there any version may compatible with that device?

kuoruan, 2019-10-12:
Which file you installed?
```
cat /etc/*-release
opkg --print-architecture
```

kkjoy, 2019-11-14:
see https://github.com/v2ray/v2ray-core/issues/843
需要开启FPU (Emulator)
一种选择是系统重新打包开启fpu。另一种是编译本软件时开启softfloat

kuoruan, 2019-11-14:
golang softfloat will auto set when FPU is not enabled in OpenWrt build.
And my ipk files for mips and mipsle enabled softfloat by default.

vongoethe, 2019-11-20:
> see [v2ray/v2ray-core#843](https://github.com/v2ray/v2ray-core/issues/843)
> 需要开启FPU (Emulator)
> 一种选择是系统重新打包开启fpu。另一种是编译本软件时开启softfloat

想请问一下编译本软件时如何开启softfloat? 我搜了半天没找到
谢谢了

vongoethe, 2019-11-20:
> see [v2ray/v2ray-core#843](https://github.com/v2ray/v2ray-core/issues/843)
> 需要开启FPU (Emulator)
> 一种选择是系统重新打包开启fpu。另一种是编译本软件时开启softfloat

我用的是arm的cpu, arm cortex A9, BCM53xx系列

nameup2, 2020-01-09:
kuoruan你好！我用github的action编译路由器固件，就是遇到一个问题，我已经把你的luci-app-v2ray git clone到编译代码里了，但是我不知道git clone你的v2ray-core该放在哪一行代码里，因为还在入门，所以很多不懂，如果有时间可以到我的github的lede目录里指导一下，帮我加一下代码吗？！我编译了几版，文件的兆数都很大，倒是别人的固件要比我小一半兆数，我也不知道问题出在哪里，因为还在学习中，希望有时间可以指导一下！万分感激！

kuoruan, 2020-01-09:
@nameup2 

```
src-git v2ray-core https://github.com/kuoruan/openwrt-v2ray.git
src-git luci-app-v2ray https://github.com/kuoruan/luci-app-v2ray.git
```

添加到 feeds.conf 或 feeds.conf.default 即可

nameup2, 2020-01-09:
> @nameup2
> 
> ```
> src-git v2ray_core https://github.com/kuoruan/openwrt-v2ray.git
> src-git v2ray_luci https://github.com/kuoruan/luci-app-v2ray.git
> ```
> 
> 添加到 feeds.conf 或 feeds.conf.default 即可
> 
> 改：OpenWrt 只认子目录，上述方法无效。
> 
> 所以你直接 clone 到 sdk 的 package 目录吧。
> 
> ```
> git clone https://github.com/kuoruan/openwrt-v2ray.git package/v2ray-core
> git clone https://github.com/kuoruan/luci-app-v2ray.git package/luci-app-v2ray
> ```

感谢大神回复，开心！我思考了一下，所以是加到下面这个代码里面对吗？
- name: 第三方插件
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages
          git clone https://github.com/kuoruan/openwrt-v2ray.git package/v2ray-core
          git clone https://github.com/kuoruan/luci-app-v2ray.git package/luci-app-v2ray

如果是以上这样的，那需要把第二行git clone https://github.com/kuoruan/luci-app-v2ray.git package/luci-app-v2ray的luci-app-v2ray复制写在下面的# 常用LuCI插件选择里吗？
          cat >> .config <<EOF
          CONFIG_PACKAGE_luci-app-accesscontrol=y
          CONFIG_PACKAGE_luci-app-adbyby-plus=y
          CONFIG_PACKAGE_luci-app-aria2=y
          CONFIG_PACKAGE_ luci-app-v2ray
          EOF

kuoruan, 2020-01-09:
直接 cd 到 SDK 根目录即可，SDK 根目录下已存在 package
```
name: 第三方插件
run: |
  cd sdk
  git clone https://github.com/kuoruan/openwrt-v2ray.git package/v2ray-core
  git clone https://github.com/kuoruan/luci-app-v2ray.git package/luci-app-v2ray
```

然后

```
cat >> .config <<EOF
CONFIG_PACKAGE_luci-app-accesscontrol=y
CONFIG_PACKAGE_luci-app-adbyby-plus=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-v2ray=y
CONFIG_PACKAGE_v2ray-core=y
EOF
```

nameup2, 2020-01-09:
> 直接 cd 到 SDK 根目录即可，SDK 根目录下已存在 package
>
> ```
> name: 第三方插件
> run: |
>   cd sdk
>   git clone https://github.com/kuoruan/openwrt-v2ray.git package/v2ray-core
>   git clone [...]

已经在编译了!明天来看下是否成功哈哈！固件大小可以用代码改变吗==！，别人编译的才我的一半大，可能是他们本地编译的效果，我研究了一下固件的这个代码，如果可以变的话，应该是在这个代码变量里面添加的，但是不知道该用什么代码范围去控制编译才不会出错
# 固件压缩:
          cat >> .config <<EOF
          CONFIG_TARGET_IMAGES_GZIP=y
          EOF

kuoruan, 2020-01-09:
```
CONFIG_V2RAY_COMPRESS_UPX=y
```

nameup2, 2020-01-09:
> ```
> CONFIG_V2RAY_COMPRESS_UPX=y
> ```

> ```
> CONFIG_V2RAY_COMPRESS_UPX=y
> ```

action失败==！提示cd; sdk; no such file or directory,要不要去我lede action里看下！

nameup2, 2020-01-09:
> > ```
> > CONFIG_V2RAY_COMPRESS_UPX=y
> > ```
> 
> > ```
> > CONFIG_V2RAY_COMPRESS_UPX=y
> > ```
> 
> action失败==！提示cd; sdk; no such file or directory,要不要去我lede action里看下！我fork的是lean大的编译代码

nameup2, 2020-01-10:
我想到一个办法，不知道可不可行，就是我先用这个代码把https://github.com/kuoruan/luci-app-v2ray.git编到自定义里面
- name: 自定义插件
        run: |
          cd package
          mkdir openwrt-packages
          cd openwrt-packages
          git clone https://github.com/kuoruan/luci-app-v2ray.git
V2RAY-core我就先编译原生自带的，这样他会编译成功出来，我试过了，成功后我再把https://github.com/kuoruan/openwrt-v2ray.git的v2ray-core 下载下来本地ssh用opkg install安装，先去实验下哈哈！

commit 5bdb6775 (2020-01-15): chore: bump to v4.22.1-4

GOARM is fixed in openwrt/packages#b36f27716f63c04d654e46dd115d5195f0c7cbba

close #33
commit a1f5c967 (2020-01-15): chore: bump to v4.22.1-4

GOARM is fixed in openwrt/packages@b36f27716f63c04d654e46dd115d5195f0c7cbba

close #33
closed 2020-01-15 by kuoruan