kuoruan/openwrt-v2ray #35: Cannot integrate precompiled ipk files with OpenWRT Image Builder (no valid architecture, ignoring)

windy32, 2019-10-26:
Working with Netgear WNDR4300v1 (ar71xx) running OpenWRT.

When trying to integrate v2ray-core_4.21.0-1_mips_24kc.ipk into the official router image, there's an error: Package v2ray-core version 4.21.0-1 has no valid architecture, ignoring.

kuoruan, 2019-10-26:
Need some information:

```
cat /usr/lib/os-release
opkg print-architecture
```

windy32, 2019-10-26:
From my router:

`BusyBox v1.23.2 (2016-01-02 21:42:02 CET) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 CHAOS CALMER (15.05.1, r48532)
 -----------------------------------------------------
  * 1 1/2 oz Gin            Shake with a glassful
  * 1/4 oz Triple Sec       of broken ice and pour
  * 3/4 oz Lime Juice       unstrained into a goblet.
  * 1 1/2 oz Orange Juice
  * 1 tsp. Grenadine Syrup
 -----------------------------------------------------
root@OpenWrt:~# cat /usr/lib/os-release
cat: can't open '/usr/lib/os-release': No such file or directory
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch ar71xx 10
`

OpenWRT Image Builder is executed on an x86-64 host. The error mentioned above is reported by OpenWRT Image Builder. Both 15.05.1 and 18.06.4 have been tested.

kuoruan, 2019-10-26:
OpenWrt renamed ar7xx to ath79: https://openwrt.org/docs/techref/targets/ath79
And my ipk files are build with the latest SDK. You need a ipk with name ar71xx.
If you want to install my ipk files, add ```arch mips_24kc 100``` to your /etc/opkg.conf

closed 2019-10-30 by kuoruan