kuoruan/openwrt-v2ray #38: using customized config file copied from /var/etc/v2ray/v2ray.main.json doesn't work

RoninYuan-WY, 2019-11-12:
I copied the whole config file from v2ray.main.json(this is a transparent proxy setting was hosted on a soft router with openwrt on X86 system, it works when using it) to file named config.json in either same fold of v2ray.main.json or /etc/v2ray/, when I tried to use customized config file with config.json in the global settings. after saving and reloading, it doesn't work. I check the config.json in the about page, it is exactly same as what is in the v2ray.main.jason. what did I missed?

kuoruan, 2019-11-12:
Transparent proxy rules will not add when you using customize config file. (Can't get the current port of the Dokodemo-door)

But next luci-app-v2ray release will allow user to set transparent proxy port.

closed 2019-11-12 by kuoruan

RoninYuan-WY, 2019-11-14:
I found the latest version still didn't work by using customized config file copied from v2ray.main.json.

kuoruan, 2019-11-14:
Need your iptables rules and config file content.

kuoruan, 2019-11-14:
You need also add your V2Ray server ip or domain to 'Extra direct list'.

RoninYuan-WY, 2019-11-14:
yes, it works finally by adding server to "Extra direct list". I just curious why this has to be done?

kuoruan, 2019-11-14:
If you don't add your server to direct list, V2Ray's output traffic will be redirect to V2Ray itself by iptables, and finally become loop.

RoninYuan-WY, 2019-11-14:
Why did not this happen when using settings from app panel? just because app handles panel settings in the different way from customized config file?

kuoruan, 2019-11-14:
Your server ip will be auto added to direct list if you use the LuCI interface.

RoninYuan-WY, 2019-11-14:
another minor problem, switching from config file to panel setting, global settings will lost original choice for panel settings last time.

kuoruan, 2019-11-14:
Yes.

hichkak, 2022-11-25:
> I copied the whole config file from v2ray.main.json(this is a transparent proxy setting was hosted on a soft router with openwrt on X86 system, it works when using it) to file named config.json in [...]

hi
What is the path for V2ray config file ?