kuoruan/openwrt-v2ray #39: 运行状态一直显示not running

sbw983, 2019-11-30:
路由器上安装完毕，已通过/etc/init.d/v2ray start启动v2ray服务，服务状态设置为enable,但是状态一直显示为not running，需要手动打开程序并制定config文件才能启动

v2ray -config /var/etc/v2ray/v2ray.main.json

kuoruan, 2019-11-30:
你是哪个版本的 LuCI？可以尝试重新安装 LuCI，由于不知道如何复现

sbw983, 2019-12-01:
LuCI openwrt-18.06 branch (git-18.219.52706-a49e4e4) / OpenWrt 18.06.1 r7258-5eb055306f

v2ray的luci端也装了，enable的选项框也点上了

kuoruan, 2019-12-02:
我指的是 luci-app-v2ray 的版本

troywall, 2019-12-14:
同遇到这个问题，luci v1.4.3-1 版本，能否出个luci配置说明?

sbw983, 2019-12-15:
一直忘记回了，好像是/etc/init.d/v2ray start不起作用，该脚本没法启动v2ray进程

sbw983, 2019-12-15:
> 我指的是 luci-app-v2ray 的版本

一直忘记回了，好像是/etc/init.d/v2ray start不起作用，该脚本没法启动v2ray进程

closed 2020-01-06 by kuoruan

orange-cat-fantacy, 2020-03-17:
cd 到etc/init.d 只有shadowsocksr  并且无法运行  但ssr的链接能运行  邪乎

kuoruan, 2020-03-17:
@orange-cat-fantacy LuCI 装好了吗？

smallshuipu, 2022-11-22:
你安装的核心是V2RAY-CORE还是MINI？MINI好像没有geosite.dat