kuoruan/openwrt-v2ray #40: ramips_24kec

mrhuangbh, 2019-12-12:
安装不上

kuoruan, 2019-12-12:
请给出必要信息：
```
cat /etc/*release
opkg print-architecture
```
没有内容的 issue 将会关闭

mrhuangbh, 2019-12-12:
root@RP-LINK:~# cat /etc/*release
DISTRIB_ID="RP-LINK"
DISTRIB_RELEASE="Bleeding Edge"
DISTRIB_REVISION="r39930"
DISTRIB_CODENAME="rpos(nt)"
DISTRIB_TARGET="ramips_new/generic"
DISTRIB_DESCRIPTION="RP-LINK RPOS(NT) r(2016-09-13 11:27:05)"
DISTRIB_TAINTS="no-all busybox"
root@RP-LINK:~# opkg print-architecture
arch all 1
arch noarch 1
arch ramips_24kec 10

kuoruan, 2019-12-13:
改一下 ```/etc/opkg.conf``` 文件，加上 mipsel_24kc 应该就行了
我这个编译都是给 OpenWrt 用的，其他平台不考虑

closed 2019-12-13 by kuoruan

sgon00, 2020-06-14:
今天在一个路由器上安装，遇到同样的问题。这是因为openwrt版本太低了，我用的是14.07，需要一个legacy库。shadowsocks提供了两个库，对于老版本，可以使用 `http://openwrt-dist.sourceforge.net/legacy/base/ramips`。不知道v2ray有没有一个类似的legacy库呢？

因为路由器设备很难更新固件，新固件对这款路由器是不支持的，所以没办法更新固件。
编译程序的话，也是问题多多，因为14.07是编译软件都是针对老系统的，在新系统上无法使用。
谢谢您。

kuoruan, 2020-06-15:
遇到这种问题请手动下载安装。

sgon00, 2020-06-15:
@kuoruan 感谢您的回复。不过直接手动下载安装也是不行的。具体原因不明，可能因为框架只认`ramips_24kec`，也可能是因为不支持14.07。我尝试了安装 `v2ray-core_4.24.2-1_mipsel_24kc.ipk`，提示错误是 `pkg_init_from_file: Malformed package file v2ray-core_4.24.2-1_mipsel_24kc.ipk.`。没有再尝试其他版本，对于直接安装ipk不报太大希望，因为没有一个后缀是 `ramips_24kec`的，这个是14.07下面的框架型号。

kuoruan, 2020-06-15:
我指的手动安装是下载官方的编译包手动拷贝到路由器上

sgon00, 2020-06-15:
@kuoruan 您指的是编译内核，编译生成ipk文件吗？如果是这个意思，我第一条信息就有说，编译14.07需要找到合适版本的操作系统，因为编译环境要求老系统，新系统下是编译不成功的，很多依赖问题。之前尝试过一次，各种报错，就放弃了。现在我在考虑其他解决方案，这台14.07打算放弃了。。。谢谢您的回复。

kuoruan, 2020-06-15:
请从这里下载官方编译包 https://github.com/v2fly/v2ray-core/releases