kuoruan/openwrt-v2ray #43: Phicomm K3 (BCM5301X) run v2ray error : Illegal instruction

wukaisky, 2020-01-08:
OpenWrt 19.07 无法运行编译完成的v2ray，运行提示：Illegal instruction

root@Phicomm-K3:~# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 0 (v7l)
BogoMIPS	: 1400.00
Features	: half thumb fastmult edsp tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

processor	: 1
model name	: ARMv7 Processor rev 0 (v7l)
BogoMIPS	: 1400.00
Features	: half thumb fastmult edsp tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x3
CPU part	: 0xc09
CPU revision	: 0

Hardware	: BCM5301X
Revision	: 0000
Serial		: 0000000000000000

kuoruan, 2020-01-08:
Related to #71
Waiting for OpenWrt to fix it.

closed 2020-01-08 by kuoruan

kuoruan, 2020-01-15:
Post the bug to OpenWrt. Seems fixed now.
openwrt/packages#10967

kuoruan, 2020-01-15:
Try 4.22.1-4

wukaisky, 2020-01-15:
Thank You. 
v2ray is now running
V2Ray 4.22.1 (OpenWrt) R3 (go1.13.6 linux/arm)