kuoruan/openwrt-v2ray #44: Can't find the package for mips64_octeon.

oemaix, 2020-01-08:
My router is ubnt erlite-3, which came with a Cavium Octeon CN5020 SOC. But v2ray-core failed to install from the package source http://openwrt.kuoruan.net/packages/releases/mips64_octeon . 
Some information is pasted as follows:

root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mips64_octeon 10

root@OpenWrt:~# cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.2'
DISTRIB_REVISION='r7676-cddd7b4c77'
DISTRIB_TARGET='octeon/generic'
DISTRIB_ARCH='mips64_octeon'
DISTRIB_DESCRIPTION='OpenWrt 18.06.2 r7676-cddd7b4c77'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="18.06.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 18.06.2"
VERSION_ID="18.06.2"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r7676-cddd7b4c77"
LEDE_BOARD="octeon/generic"
LEDE_ARCH="mips64_octeon"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt 18.06.2 r7676-cddd7b4c77"

kuoruan, 2020-01-08:
Currently has no build for your arch.
I will add mips64_octeon later

closed 2020-01-08 by kuoruan

kuoruan, 2020-01-09:
mips64_octeon is renamed to mips64_octeonplus in OpenWrt new versions.

commit 5f059e30 (2020-01-08): ci: add more build targets (#44)

close #44