kuoruan/openwrt-v2ray #47: 请问这种情况会是客户端的问题吗？

Totuz, 2020-01-16:
```
> cat /etc/*release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01-SNAPSHOT'
DISTRIB_REVISION='r4076-42f7ad1'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='ipq806x/generic'
DISTRIB_ARCH='arm_cortex-a15_neon-vfpv4'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01-SNAPSHOT r4076-42f7ad1'
DISTRIB_TAINTS='no-all busybox'
NAME="LEDE"
VERSION="17.01-SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot 17.01-SNAPSHOT"
VERSION_ID="17.01-snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4076-42f7ad1"
LEDE_BOARD="ipq806x/generic"
LEDE_ARCH="arm_cortex-a15_neon-vfpv4"
LEDE_TAINTS="no-all busybox"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot 17.01-SNAPSHOT r4076-42f7ad1"

>opkg print-architecture
arch all 1
arch noarch 1
arch arm_cortex-a15_neon-vfpv4 10
```

感谢你的劳动与分享。
因为被难住太久，搜索没能解决，所以来这边求助。
我用的release是 v2ray-core_4.22.1-4_arm_cortex-a15_neon-vfpv4.ipk
手机和电脑上的v2ray使用相同服务器没问题，路由器上1.1.1.1 DNS转发也正常
附上截图说明一下我遇到的问题：

[image]
客户端

[image]
服务端

firewall规则
```
# ipset -R < /etc/config/v2ray/ad.ips
ipset -R < /etc/config/v2ray/gw.ips

# resever IP
iptables -t mangle -I PREROUTING -d 127.0.0.0/24 -j RETURN
iptables -t mangle -I PREROUTING -d 192.168.0.0/16 -j RETURN
iptables -t mangle -I PREROUTING -d 10.42.0.0/16 -j RETURN
iptables -t mangle -I PREROUTING -d 0.0.0.0/8 -j RETURN
iptables -t mangle -I PREROUTING -d 10.0.0.0/8 -j RETURN
iptables -t mangle -I PREROUTING -d 172.16.0.0/12 -j RETURN
iptables -t mangle -I PREROUTING -d 224.0.0.0/4 -j RETURN
iptables -t mangle -I PREROUTING -d 240.0.0.0/4 -j RETURN
iptables -t mangle -I PREROUTING -d 169.254.0.0/16 -j RETURN
iptables -t mangle -I PREROUTING -d 255.255.0.0/8 -j RETURN

ip rule add fwmark 0x01/0x01 table 100
ip route add local 0.0.0.0/0 dev lo table 100
# iptables -t nat -A PREROUTING -m set --match-set ad dst -j REJECT
iptables -t mangle -A PREROUTING -p tcp -m set --match-set gw dst -j TPROXY --on-port 12345 --tproxy-mark 0x01/0x01
iptables -t mangle -A PREROUTING -p udp -m set --match-set gw dst -j TPROXY --on-port 12345 --tproxy-mark 0x01/0x01
iptables -t mangle -A PREROUTING -p udp -d 1.1.1.1 -j TPROXY --on-port 12345 --tproxy-mark 0x01/0x01
iptables -t mangle -A OUTPUT -m mark --mark 2 -j RETURN
iptables -t mangle -A OUTPUT -p udp -d 1.1.1.1 -j MARK --set-mark 0x01/0x01

# The following two lines are for router self (optional)

iptables -t mangle -A OUTPUT -p tcp -m set --match-set gw dst -j MARK --set-mark 0x01/0x01
iptables -t mangle -A OUTPUT -p udp -m set --match-set gw dst -j MARK --set-mark 0x01/0x01
```

客户端config
```
{
  "log" : 
  {
    "access": "/var/log/v2ray/access.log",
    "error": "/var/log/v2ray/error.log",
    "loglevel": "debug"
  },
  "inbounds": [
  {
    "listen": "0.0.0.0",
    "port": 12345,
    "protocol": "dokodemo-door",
    "settings": {
      "network": "tcp,udp",
      "followRedirect": true
    },
    "streamSettings": {
      "sockopt": {
      "tproxy": "tproxy"
      }
     }
  }
  ],
  "outbounds": [
  {
    "protocol": "vmess",
    "tag": "proxy",
    "settings": {
      "vnext": [{
        "address": "servername",
        "port": 443,
        "users": [{
          "id": "uuid",
          "alterId": 64,
          "level": 1,
          "security": "aes-128-gcm"
        }]
      }]
    },
    "streamSettings": {
      "network" : "ws",
      "security": "tls",
      "wsSettings": {
        "path": "/path"
      },
      "sockopt": {
        "mark": 2,
        "tproxy": "tproxy"
      },
      "tlsSettings": {
        "serverName": "servername",
        "alpn" : [
            "http\/1.2"
          ],
        "allowInsecure": false,
        "allowInsecureCiphers" : false
      }
    }
  }
  ]
}
```

服务端config
```
{
  "log" :
  {
    "loglevel" : "debug",
    "access" : "/var/log/v2ray/access.log",
    "error" : "/var/log/v2ray/error.log"
  },

  "inbound":
  {
    "port": 12345,
    "protocol": "vmess",

    "settings": {
      "clients": [
        {
          "id": "uuid",
          "level": 1,
          "alterId": 64
        }
      ]
    },

   "streamSettings": {
   "network": "ws",
   "wsSettings": {
      "path": "/path"
      }
    }
  },

  "outbound":
  {
    "protocol": "freedom",
    "settings": {}
  },

  "outboundDetour": [
    {
      "protocol": "blackhole",
      "settings": {},
      "tag": "blocked"
    }
  ],

  "routing":
  {
    "strategy": "rules",
    "settings": {
      "rules": [
        {
          "type": "field",
          "ip": ["geoip:private"],
          "outboundTag": "blocked"
        }
      ]
    }
  }
}
```

kuoruan, 2020-01-17:
规则上没看出来什么问题，唯一不同的是客户端和服务端的 security 项，不确定是否有影响

kuoruan, 2020-01-17:
我对 V2Ray 的内部运行机制不熟，你可以去官方开个 issue 问问

Totuz, 2020-01-17:
> 我对 V2Ray 的内部运行机制不熟，你可以去官方开个 issue 问问

security是服务器自适应没问题的。
感谢你的回复，我去v2ray再问一下。

closed 2020-01-17 by Totuz