kuoruan/openwrt-v2ray #48: `/etc/init.d/v2ray stop` reboots the router

poscat0x04, 2020-01-21:
Please attach the output of following commands.

```
cat /etc/*release

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.0'
DISTRIB_REVISION='r10860-a3ffeb413b'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='OpenWrt 19.07.0 r10860-a3ffeb413b'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.0"
VERSION_ID="19.07.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r10860-a3ffeb413b"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.0 r10860-a3ffeb413b

opkg print-architecture

arch all 1
arch noarch 1
arch arm_cortex-a9_vfpv3 10
```

kuoruan, 2020-01-22:
Can't reproduce. Firewall may restart when v2ray stop is called. but we do not reboot the router.

imgk, 2020-01-26:
I also have this problem when stopping v2ray.

poscat0x04, 2020-02-05:
The issue is gone after I updated all the packages. I suspect it was caused by some bugs in the dependencies. Closing.

closed 2020-02-05 by poscat0x04