kuoruan/openwrt-v2ray #49: 启用服务导致 ER-X 重启长时间无法连接

mrhso, 2020-01-27:
环境：OpenWrt 18.06.5 on UBNT ER-X
版本：core v4.22.1-4，luci-app v1.5.3-1
表现为启动后一段时间（不定），路由器无法与外部端口通信。
若在 LuCI 中勾选「已启用」，重启后会触发问题，反之不会。
附系统日志：
```
Tue Jan 28 01:02:56 2020 kern.notice kernel: [    0.000000] Linux version 4.14.151 (buildbot@b87496cfc8a9) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7897-9d401013fc)) #0 SMP Tue Nov 5 14:12:18 2019
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] MIPS: machine is UBNT-ERX
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Determined physical RAM map:
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000]  memory: 10000000 @ 00000000 (usable)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Zone ranges:
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000]   HighMem  empty
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Early memory node ranges
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 80556140, node_mem_map 81003000
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000]   Normal zone: 512 pages used for memmap
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Tue Jan 28 01:02:56 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from 0x80559744 with crng_init=0
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] percpu: Embedded 14 pages/cpu s26224 r8192 d22928 u57344
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s26224 r8192 d22928 u57344 alloc=14*4096
Tue Jan 28 01:02:56 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
Tue Jan 28 01:02:56 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00048cc0
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00048cc0
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Memory: 253620K/262144K available (4648K kernel code, 243K rwdata, 572K rodata, 220K init, 254K bss, 8524K reserved, 0K cma-reserved, 0K highmem)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] Hierarchical RCU implementation.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.015497] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.087820] pid_max: default: 32768 minimum: 301
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.097137] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.110158] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.125892] Hierarchical SRCU implementation.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.135361] smp: Bringing up secondary CPUs ...
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.145069] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.145077] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.145088] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.145220] CPU1 revision is: 0001992f (MIPS 1004Kc)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.204724] Synchronize counters for CPU 1: done.
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.268841] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.268849] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.268857] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.268929] CPU2 revision is: 0001992f (MIPS 1004Kc)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.325575] Synchronize counters for CPU 2: done.
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.385922] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.385930] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Tue Jan 28 01:02:56 2020 kern.warn kernel: [    0.385938] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.386028] CPU3 revision is: 0001992f (MIPS 1004Kc)
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.445140] Synchronize counters for CPU 3: done.
Tue Jan 28 01:02:56 2020 kern.info kernel: [    0.504741] smp: Brought up 1 node, 4 CPUs
[... 212 lines cut ...]
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'lan' is enabled
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'lan' is setting up now
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'lan' is now up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'loopback' is enabled
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'loopback' is setting up now
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'loopback' is now up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'wan' is enabled
Tue Jan 28 01:48:13 2020 daemon.notice netifd: bridge 'br-lan' link is up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Network device 'eth0' link is up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: VLAN 'eth0.1' link is up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Network device 'lo' link is up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'loopback' has link connectivity
Tue Jan 28 01:48:13 2020 daemon.notice netifd: VLAN 'eth0.2' link is up
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Jan 28 01:48:13 2020 daemon.notice netifd: Interface 'wan' is setting up now
[... cut ...]
Tue Jan 28 01:48:13 2020 daemon.err insmod: module is already loaded - pppox
Tue Jan 28 01:48:13 2020 daemon.err insmod: module is already loaded - pppoe
Tue Jan 28 01:48:13 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Jan 28 01:48:14 2020 daemon.info pppd[1101]: Plugin rp-pppoe.so loaded.
Tue Jan 28 01:48:14 2020 daemon.info pppd[1101]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Tue Jan 28 01:48:14 2020 daemon.info dnsmasq[735]: exiting on receipt of SIGTERM
```

kuoruan, 2020-01-28:
看你的防火墙情况。比较一下正常情况下的防火墙规则和不正常情况下的防火墙规则

mrhso, 2020-01-28:
开启前后防火墙规则未见变化。另外 `random: crng init done` 明显滞后，在此之前均处于无法连接的状态。

kuoruan, 2020-01-28:
重启时间长是因为需要等待 dnsmasq 载入规则列表，这个时间比较长

mrhso, 2020-01-28:
> 重启时间长是因为需要等待 dnsmasq 载入规则列表，这个时间比较长

45 分钟的话似乎不太正常……
另外我用了自定义配置，不知道有没有关系。

kuoruan, 2020-01-28:
路由什么架构的？

mrhso, 2020-01-28:
mipsel_24kc MT7621

kuoruan, 2020-01-29:
重启之前把 ```/tmp/dnsmasq.d/v2ray``` 删了试试，或者不要启用 GFWList

mrhso, 2020-01-29:
启用 V2Ray 之后 `/tmp/dnsmasq.d/` 是空的。

kuoruan, 2020-01-30:
不能啊，那你这些日志怎么来的？
```
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain www.gstatic.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain www-googletagmanager.l.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain update.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain translate.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain tools.l.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain tools.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain toolbarqueries.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain ssl.gstatic.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain ssl-google-analytics.l.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain settings.crashlytics.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain safebrowsing.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain safebrowsing-cache.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain redirector.gvt1.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain recaptcha.net
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain pagead-googlehosted.l.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain ocsp.pki.goog
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain media.admob.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khmdb.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khmdb.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm3.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm3.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm2.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm2.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm1.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm1.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm0.googleapis.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm0.google.com
Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain khm.googleapis.com
Tue Jan 28 01:03:00 2020 daemon.info dnsmasq[735]: using 73431 more nameservers
```

mrhso, 2020-01-30:
> 不能啊，那你这些日志怎么来的？
>
> ```
> Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using nameserver 114.114.114.114#53 for domain www.gstatic.com
> Tue Jan 28 01:02:59 2020 daemon.info dnsmasq[735]: using [...]

那是我自己给 dnsmasq 配置的，而且禁用 V2Ray 之后这些也并没有显著拖慢启动。

kuoruan, 2020-01-30:
还有就是你这个 CPU 太弱了，本来启动 v2ray 就很慢。

nobk, 2020-05-03:
这个问题我也发现，并且已经解决了，不是v2ray的问题，而是luci-app-v2ray的init.d启动脚本被防火墙规则提前reload造成的，按我贴出的方式改一下脚本，路由器就能恢复正常重启时间了。
https://github.com/kuoruan/luci-app-v2ray/issues/141

nobk, 2020-05-03:
> 还有就是你这个 CPU 太弱了，本来启动 v2ray 就很慢。

我也是MT7621，是最高版本MT7621A，双核四线程的，没FPU用软件模拟协处理器做浮点运算，跑V2RAY也能流畅看油管4K视频，外网走代理下载速度能到5MB/s。而且这个硬件平台的路由器特省电，无线只开5G才3W左右，我这里停电后直流UPS也能带机22小时。