kuoruan/openwrt-v2ray #50: 放置一夜无法访问国内，国外正常，重启依然

macdoor, 2020-02-13:
睡觉前一切正常，放置一夜无法访问国内，国外正常，重启依然，连续2天如此

```
版本信息
主机名	OpenWrt
型号	Raspberry Pi 3 Model B Rev 1.2
架构	?
固件版本	OpenWrt 19.07.1 r10911-c155900f66 / LuCI openwrt-19.07 branch git-20.029.45734-adbbd5c
内核版本	4.14.167

版本：V2Ray 4.22.1 (OpenWrt) R4 (go1.13.6 linux/arm64)
```

睡觉前 ping taobao.com 可以通，现在不通了
截取了部分访问taobao.com 的日志

```
2020/02/13 04:09:36 [Info] [747188646] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.111:8555
2020/02/13 04:09:37 [Info] v2ray.com/core/app/dns: UDP:1.1.1.1:53 got answere: clients1.google.com. TypeA -> [172.217.5.206] 259.27677ms
2020/02/13 04:09:37 [Info] v2ray.com/core/app/dns: UDP:1.1.1.1:53 got answere: clients1.google.com. TypeAAAA -> [[2607:f8b0:4007:80d::200e]] 259.361562ms
2020/02/13 04:09:37 [Info] [4133227951] v2ray.com/core/app/dispatcher: default route for tcp:clients1.google.com:443
2020/02/13 04:09:37 [Info] v2ray.com/core/transport/internet/kcp: dialing mKCP to udp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:37 [Info] v2ray.com/core/transport/internet/kcp: #20793 creating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:37 [Info] [4133227951] v2ray.com/core/proxy/vmess/outbound: tunneling request to tcp:clients1.google.com:443 via tcp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:37 [Info] v2ray.com/core/transport/internet/kcp: #20738 terminating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:38 [Info] v2ray.com/core/transport/internet/kcp: #20788 terminating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:38 [Info] v2ray.com/core/transport/internet/kcp: #20789 terminating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:39 [Info] [1245172736] v2ray.com/core/app/dispatcher: sniffed domain: radio.itunes.apple.com
2020/02/13 04:09:39 [Info] v2ray.com/core/app/dns: UDP:1.1.1.1:53 got answere: radio.itunes.apple.com. TypeA -> [104.70.228.76] 233.087083ms
2020/02/13 04:09:39 [Info] [1245172736] v2ray.com/core/app/dispatcher: default route for tcp:radio.itunes.apple.com:443
2020/02/13 04:09:39 [Info] v2ray.com/core/transport/internet/kcp: dialing mKCP to udp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:39 [Info] v2ray.com/core/transport/internet/kcp: #20794 creating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:39 [Info] [1245172736] v2ray.com/core/proxy/vmess/outbound: tunneling request to tcp:radio.itunes.apple.com:443 via tcp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:39 [Info] v2ray.com/core/app/dns: UDP:1.1.1.1:53 got answere: radio.itunes.apple.com. TypeAAAA -> [[2600:1406:5800:58e::2a1] [2600:1406:5800:5aa::2a1] [2600:1406:5800:5b0::2a1] [2600:1406:5800:5bc::2a1] [2600:1406:5800:586::2a1]] 234.011927ms
2020/02/13 04:09:40 [Info] [2384138763] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.110:8555
2020/02/13 04:09:40 [Info] [747188646] v2ray.com/core/app/proxyman/outbound: failed to process outbound traffic > v2ray.com/core/proxy/freedom: failed to open connection to tcp:123.123.190.111:8555 > v2ray.com/core/common/retry: [dial tcp 123.123.190.111:8555: connect: no route to host] > v2ray.com/core/common/retry: all retry attempts failed
2020/02/13 04:09:40 [Info] [747188646] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/dokodemo: connection ends > v2ray.com/core/proxy/dokodemo: failed to transport response > io: read/write on closed pipe
2020/02/13 04:09:40 [Info] v2ray.com/core/transport/internet/kcp: #20794 closing connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:40 [Info] v2ray.com/core/transport/internet/kcp: #20783 terminating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:43 [Info] [2384138763] v2ray.com/core/app/proxyman/outbound: failed to process outbound traffic > v2ray.com/core/proxy/freedom: failed to open connection to tcp:123.123.190.110:8555 > v2ray.com/core/common/retry: [dial tcp 123.123.190.110:8555: connect: no route to host] > v2ray.com/core/common/retry: all retry attempts failed
2020/02/13 04:09:43 [Info] [2384138763] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/dokodemo: connection ends > v2ray.com/core/proxy/dokodemo: failed to transport response > io: read/write on closed pipe
2020/02/13 04:09:44 [Info] v2ray.com/core/transport/internet/kcp: #20791 terminating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:45 [Info] [4034608443] v2ray.com/core/app/dispatcher: taking detour [direct] for [tcp:123.123.190.111:8555]
2020/02/13 04:09:45 [Info] [4034608443] v2ray.com/core/proxy/freedom: opening connection to tcp:123.123.190.111:8555
2020/02/13 04:09:45 [Info] [4034608443] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.111:8555
2020/02/13 04:09:45 [Info] v2ray.com/core/transport/internet/kcp: #20793 closing connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:45 [Info] [804493368] v2ray.com/core/app/dispatcher: sniffed domain: www.taobao.com
2020/02/13 04:09:45 [Info] v2ray.com/core/transport/internet/udp: establishing new connection for udp:114.114.114.114:53
2020/02/13 04:09:45 [Info] v2ray.com/core/app/dispatcher: taking detour [direct] for [udp:114.114.114.114:53]
2020/02/13 04:09:45 [Info] v2ray.com/core/proxy/freedom: opening connection to udp:114.114.114.114:53
2020/02/13 04:09:45 [Info] [4034608443] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.111:8555
2020/02/13 04:09:46 [Info] v2ray.com/core/transport/internet/kcp: #20790 closing connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:46 [Info] v2ray.com/core/app/proxyman/outbound: failed to process outbound traffic > v2ray.com/core/proxy/vmess/outbound: connection ends > context canceled
2020/02/13 04:09:46 [Info] v2ray.com/core/transport/internet/udp: failed to handle UDP input > io: read/write on closed pipe
2020/02/13 04:09:47 [Info] v2ray.com/core/transport/internet/kcp: #20793 terminating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:48 [Info] [412316063] v2ray.com/core/app/dispatcher: taking detour [direct] for [tcp:123.123.190.110:8555]
2020/02/13 04:09:48 [Info] [412316063] v2ray.com/core/proxy/freedom: opening connection to tcp:123.123.190.110:8555
2020/02/13 04:09:48 [Info] [412316063] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.110:8555
2020/02/13 04:09:49 [Info] [412316063] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.110:8555
2020/02/13 04:09:49 [Info] [4034608443] v2ray.com/core/transport/internet/tcp: dialing TCP to tcp:123.123.190.111:8555
2020/02/13 04:09:49 [Info] v2ray.com/core/app/dns: failed to lookup ip for domain www.taobao.com at server UDP:114.114.114.114:53 > context deadline exceeded
2020/02/13 04:09:49 [Info] v2ray.com/core/transport/internet/udp: establishing new connection for udp:1.1.1.1:53
2020/02/13 04:09:49 [Info] v2ray.com/core/app/dispatcher: taking detour [proxy] for [udp:1.1.1.1:53]
2020/02/13 04:09:49 [Info] v2ray.com/core/transport/internet/kcp: dialing mKCP to udp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:49 [Info] v2ray.com/core/app/proxyman/outbound: failed to process outbound traffic > v2ray.com/core/proxy/freedom: connection ends > context canceled
2020/02/13 04:09:49 [Info] v2ray.com/core/transport/internet/udp: failed to handle UDP input > io: read/write on closed pipe
2020/02/13 04:09:49 [Info] v2ray.com/core/transport/internet/kcp: #20795 creating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:49 [Info] v2ray.com/core/proxy/vmess/outbound: tunneling request to udp:1.1.1.1:53 via tcp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:09:50 [Info] v2ray.com/core/app/dns: UDP:1.1.1.1:53 got answere: www.taobao.com. TypeA -> [47.246.25.233 47.246.25.234] 223.423125ms
2020/02/13 04:09:50 [Info] [804493368] v2ray.com/core/app/dispatcher: taking detour [direct] for [tcp:www.taobao.com:443]
[... 106 lines cut ...]
2020/02/13 04:10:05 [Info] v2ray.com/core/transport/internet/kcp: #20797 creating connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:10:05 [Info] [726102325] v2ray.com/core/proxy/vmess/outbound: tunneling request to tcp:jd.sital.space:80 via tcp:[xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:10:05 [Info] v2ray.com/core/transport/internet/kcp: #20795 closing connection to [xxxx:xxxx:1:1e9::xxxx]:xxxx
2020/02/13 04:10:05 [Info] v2ray.com/core/app/proxyman/outbound: failed to process outbound traffic > v2ray.com/core/proxy/vmess/outbound: connection ends > context canceled
[... cut ...]
2020/02/13 04:10:08 [Info] [1863397642] v2ray.com/core/app/proxyman/inbound: connection ends > v2ray.com/core/proxy/dokodemo: connection ends > v2ray.com/core/proxy/dokodemo: failed to transport response > io: read/write on closed pipe
2020/02/13 04:10:08 [Info] [1268938269] v2ray.com/core/app/dispatcher: sniffed domain: count.taobao.com
2020/02/13 04:10:08 [Info] v2ray.com/core/transport/internet/udp: establishing new connection for udp:114.114.114.114:53
2020/02/13 04:10:08 [Info] v2ray.com/core/app/dispatcher: taking detour [direct] for [udp:114.114.114.114:53]
2020/02/13 04:10:08 [Info] v2ray.com/core/proxy/freedom: opening connection to udp:114.114.114.114:53

```

kuoruan, 2020-02-13:
防火墙规则贴一下

kuoruan, 2020-02-13:
可以尝试添加一条路由，让 114.114.114.114 走 direct

macdoor, 2020-02-13:
```
root@OpenWrt:~# iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere             /* !fw3 */
input_rule  all  --  anywhere             anywhere             /* !fw3: Custom input rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
syn_flood  tcp  --  anywhere             anywhere             tcp flags:FIN,SYN,RST,ACK/SYN /* !fw3 */
zone_lan_input  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_input  all  --  anywhere             anywhere             /* !fw3 */

Chain FORWARD (policy DROP)
target     prot opt source               destination         
forwarding_rule  all  --  anywhere             anywhere             /* !fw3: Custom forwarding rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
zone_lan_forward  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_forward  all  --  anywhere             anywhere             /* !fw3 */
reject     all  --  anywhere             anywhere             /* !fw3 */

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere             /* !fw3 */
output_rule  all  --  anywhere             anywhere             /* !fw3: Custom output rule chain */
ACCEPT     all  --  anywhere             anywhere             ctstate RELATED,ESTABLISHED /* !fw3 */
zone_lan_output  all  --  anywhere             anywhere             /* !fw3 */
zone_wan_output  all  --  anywhere             anywhere             /* !fw3 */

Chain forwarding_lan_rule (1 references)
target     prot opt source               destination         

Chain forwarding_rule (1 references)
target     prot opt source               destination         

Chain forwarding_wan_rule (1 references)
target     prot opt source               destination         

Chain input_lan_rule (1 references)
target     prot opt source               destination         

Chain input_rule (1 references)
target     prot opt source               destination         

Chain input_wan_rule (1 references)
target     prot opt source               destination         

Chain output_lan_rule (1 references)
target     prot opt source               destination         

Chain output_rule (1 references)
target     prot opt source               destination         

Chain output_wan_rule (1 references)
target     prot opt source               destination         

Chain reject (3 references)
target     prot opt source               destination         
REJECT     tcp  --  anywhere             anywhere             /* !fw3 */ reject-with tcp-reset
REJECT     all  --  anywhere             anywhere             /* !fw3 */ reject-with icmp-port-unreachable

Chain syn_flood (1 references)
target     prot opt source               destination         
RETURN     tcp  --  anywhere             anywhere             tcp flags:FIN,SYN,RST,ACK/SYN limit: avg 25/sec burst 50 /* !fw3 */
DROP       all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_dest_ACCEPT (4 references)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             anywhere             /* !fw3 */

Chain zone_lan_forward (1 references)
target     prot opt source               destination         
[... 31 lines cut ...]
forwarding_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan forwarding rule chain */
zone_lan_dest_ACCEPT  esp  --  anywhere             anywhere             /* !fw3: Allow-IPSec-ESP */
zone_lan_dest_ACCEPT  udp  --  anywhere             anywhere             udp dpt:isakmp /* !fw3: Allow-ISAKMP */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port forwards */
zone_wan_dest_REJECT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_input (1 references)
target     prot opt source               destination         
input_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan input rule chain */
ACCEPT     udp  --  anywhere             anywhere             udp dpt:bootpc /* !fw3: Allow-DHCP-Renew */
ACCEPT     icmp --  anywhere             anywhere             icmp echo-request /* !fw3: Allow-Ping */
ACCEPT     igmp --  anywhere             anywhere             /* !fw3: Allow-IGMP */
ACCEPT     all  --  anywhere             anywhere             ctstate DNAT /* !fw3: Accept port redirections */
zone_wan_src_REJECT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_output (1 references)
target     prot opt source               destination         
output_wan_rule  all  --  anywhere             anywhere             /* !fw3: Custom wan output rule chain */
zone_wan_dest_ACCEPT  all  --  anywhere             anywhere             /* !fw3 */

Chain zone_wan_src_REJECT (1 references)
target     prot opt source               destination         
reject     all  --  anywhere             anywhere             /* !fw3 */

```

macdoor, 2020-02-13:
> 可以尝试添加一条路由，让 114.114.114.114 走 direct

iptables 完全不会搞，帮忙给个命令

macdoor, 2020-02-13:
```
root@OpenWrt:~# more /etc/config/firewall
config defaults
	option syn_flood '1'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'

config zone
	option name 'lan'
	list network 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'

config zone
	option name 'wan'
	list network 'wan'
	list network 'wan6'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option src_ip 'fc00::/6'
	option dest_ip 'fc00::/6'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	list icmp_type '143/0'

config defaults
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config include
	option path '/etc/firewall.user'

config include 'v2ray'
	option type 'script'
	option path '/etc/firewall.v2ray'
	option family 'any'
	option reload '1'

```

kuoruan, 2020-02-13:
```
2020/02/13 04:09:55 [Info] v2ray.com/core/app/dns: failed to lookup ip for domain ald.taobao.com at server UDP:114.114.114.114:53 > context deadline exceeded
2020/02/13 04:09:55 [Info] v2ray.com/core/app/dns: failed to lookup ip for domain tce.taobao.com at server UDP:114.114.114.114:53 > context deadline exceeded
```

这不是能看出来是因为 dns 查询失败吗，在 v2ray 里边加个路由，让 114.114.114.114 走 direct。
或者再配置一个国内的 dns 服务器到 v2ray 里边

macdoor, 2020-02-13:
现在ping 114.114.114.114不通，但同可以curl google.com，说明网络没有问题，同时我通过主路由可以 ping 114.114.114.114 是通的。

我并不确定是v2ray的问题，也有可能openwrt 19.07问题，但肯定是有问题了。

```
root@OpenWrt:/tmp/sysinfo# ping 114.114.114.114
PING 114.114.114.114 (114.114.114.114): 56 data bytes
^C
--- 114.114.114.114 ping statistics ---
54 packets transmitted, 0 packets received, 100% packet loss
root@OpenWrt:/tmp/sysinfo# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
74 packets transmitted, 0 packets received, 100% packet loss
root@OpenWrt:/tmp/sysinfo# curl -x socks5://127.0.0.1:1080 https://google.com
<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="https://www.google.com/">here</A>.
</BODY></HTML>
root@OpenWrt:/tmp/sysinfo# 

```

macdoor, 2020-02-13:
稍等，我又重启了一次，这次ping 114.114.114.114 通了，也能翻了，再放一段时间试试看看

macdoor, 2020-02-14:
今天上午出现同样问题，国外通，国内不通，ping 114.114.114.114 不通，我现在怀疑是不是 OpenWrt 19.07.1 自身存在问题，我reset OpenWrt 回出厂设置，不安装 v2ray 相关软件，进行对比

macdoor, 2020-02-17:
确认不开启v2ray，使用 openwrt 一段时间也会产生 ping 114.114.114.114 不通的现象，我再查查，先关闭问题

closed 2020-02-17 by macdoor