kuoruan/openwrt-v2ray #51: 大佬可以支持32位的PowerPC构架处理器吗！

wazzx, 2020-02-21:
目前openwrt只有少数的基于PowerPC的处理器的路由器
网件4700就是其中之一。希望得到支持。用来大佬的编译以后出现了如下问题
Finding targets
bash: go: command not found

Building targets
bash: go: command not found

Cleaning module download cache (golang/go#27455)
bash: go: command not found

Makefile:303: recipe for target '/home/openwrt-sdk/build_dir/target-powerpc_464fp_musl/v2ray-core-4.22.1/.built' failed
make[2]: *** [/home/openwrt-sdk/build_dir/target-powerpc_464fp_musl/v2ray-core-4.22.1/.built] Error 127
make[2]: Leaving directory '/home/openwrt-sdk/package/v2ray'
time: package/v2ray/compile#2.90#0.85#12.62
package/Makefile:111: recipe for target 'package/v2ray/compile' failed
make[1]: *** [package/v2ray/compile] Error 2
make[1]: Leaving directory '/home/openwrt-sdk'
/home/openwrt-sdk/include/toplevel.mk:218: recipe for target 'package/v2ray/compile' failed
make: *** [package/v2ray/compile] Error 2

编译用的SDK：openwrt-sdk-19.07.1-apm821xx-nand_gcc-7.5.0_musl.Linux-x86_64
[WNDR4700]# cat /proc/cpuinfo 
processor       : 0
cpu             : APM82181
clock           : 1000.000010MHz
revision        : 28.131 (pvr 12c4 1c83)
bogomips        : 2000.00
timebase        : 1000000010
platform        : PowerPC 44x Platform
model           : amcc,wdnr4700
Memory          : 256 MB

CPU:   AMCC PowerPC APM82181 Rev. E at 1000 MHz (PLB=166, OPB=83, EBC=83 MHz)
       Security support
       Bootstrap Option D - Boot ROM Location NAND wo/ECC 2k page (8 bits), booting from NAND
       32 kB I-Cache 32 kB D-Cache
DNI HW ID: 29763875 NAND flash 128MB RAM 256MB U-Boot 2009.08 dni1 V1.4
Board: AMCC APM82181 Evaluation Board, PCIE0/SATA1, 1*USB OTG
I2C:   ready
DRAM:  256 MB (ECC not enabled, 500 Mb/s, 32-bit, CL3)
NAND:  128 MiB
*** Warning - bad CRC, using default environment

Configure Max Payload 256B
PCIE0: successfully set as root-complex
        03  00  168c  0030  0280  ff
        02  02  111d  803a  0604  00
        04  00  168c  0033  0280  ff
        02  03  111d  803a  0604  00
        05  00  1912  0015  0c03  00
        02  04  111d  803a  0604  00
        01  00  111d  803a  0604  00
 131072 bytes read: OK
Net:   ppc_4xx_eth0
AR8327 GigaSW initialize...done
ENET Speed is 1000 Mbps - FULL duplex connection (EMAC0)

 Client starts...[Listening] for ADVERTISE...TTT
Retry count exceeded; boot the image as usual

 nmrp server is stopped or failed !

Type run flash_nfs to mount root filesystem over NFS

Hit any key to stop autoboot:  0

kuoruan, 2020-02-21:
golang 还官方不支持 golang/go#22885

closed 2020-02-21 by kuoruan

wazzx, 2020-02-22:
大佬可以支持gccgo-go交叉编译吗！