kuoruan/openwrt-v2ray #52: 希望可以开启UPX压缩编译

wazzx, 2020-02-29:
你好K大谢谢你的发布，可以在mips构架下默认开启UPX压缩吗，目前很多mips构架下的ROM只有32M左右，所以目前12M的大小不方便直接update和install安装，部分虽然有128M的nand，但是官方uboot为了稳定和避免出现坏道一般实际可用的大概也就30M左右，我这边用自己的极路由b70因为是官方默认uboot的问题实际可以大小也即22m，只能自己编译upx压缩后（只有8m大小）然后上传安装。ARM的也有同样的问题。谢谢

kuoruan, 2020-02-29:
您好，可以自行编译

wazzx, 2020-03-03:
K大谢谢你的回复，我自己编译是没有问题，主要是想偷懒。。。。因为是在vps上面编译，所以想直接使用的源，直接opkg update 然后install，谢谢啊！

wazzx, 2020-03-10:
还是需要upx压缩的，用k大的源主要就是方便opkg update一次合成，但是限制于mips系列构架ROM普遍的32MB左右，所以还是有压力的。谢谢

shmily103, 2020-04-10:
> 您好，可以自行编译

求教一下，Project V里面选中了upx编译会报错（tools里已经增加了upx），请问需要修改代码吗？

kuoruan, 2020-04-10:
请给报错信息！

shmily103, 2020-04-11:
> 请给报错信息！

没有问题了，之前忘记执行 make clean 。
不过压缩效率一般，编译出来还有9.1M。

kuoruan, 2020-04-12:
> DO NOT BUY DEVICES WITH 4MB FLASH / 32MB RAM if you intend to flash an up-to-date and secure OpenWrt version (18.06 or later) onto it! See 4/32 warning for details.

https://openwrt.org/supported_devices/432_warning

closed 2020-04-12 by kuoruan

nobk, 2020-05-03:
由于ROM的文件系统squashfs本身就是带压缩的，再用upx压一遍除了拖慢启动速度，空间占用的改进并不大，我就直接打包进固件没做压缩。可以看到我的/rom的文件总空间占用是87.9MB，然而/rom本身的物理大小才25.8MB。
zsh#root@OpenWrt /rom
df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                25.8M     25.8M         0 100% /rom
tmpfs                   123.8M      1.5M    122.2M   1% /tmp
/dev/mmcblk0p1          487.9M    122.3M    329.8M  27% /overlay
overlayfs:/overlay      487.9M    122.3M    329.8M  27% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk0p2            3.9G    139.7M      3.5G   4% /overlay-ext4
/dev/mtdblock6            4.2M    804.0K      3.4M  19% /overlay-boot
tmpfs                   123.8M      1.5M    122.2M   1% /www/tmp
zsh#root@OpenWrt /rom
du . -m -h | tail
20.5K   ./www/luci-static/resources/view/status/include
73.0K   ./www/luci-static/resources/view/status
3.0K    ./www/luci-static/resources/view/system/led-trigger
54.5K   ./www/luci-static/resources/view/system
5.5K    ./www/luci-static/resources/view/upnp
357.5K  ./www/luci-static/resources/view
713.0K  ./www/luci-static/resources
782.5K  ./www/luci-static
795.5K  ./www
87.9M   .
zsh#root@OpenWrt /rom
mount | grep rom
/dev/root on /rom type squashfs (ro,relatime)