kuoruan/openwrt-v2ray #54: Luci 太复杂了

DanielChen008, 2020-03-03:
我是做产品和UX的，冒昧的提个想法，对于普通用户来说最佳的用户体验：
1：配置管理界面，允许添加自己的 v2ray 配置信息；
2：全局设置界面，允许勾选启用，代理模式是：全局、大陆白名单、GFWList模式 即可。

DNS 交给像 ChinaDNS 或者 SmartDNS 解决。

如果能有个精简版的 Luci 就太棒了！

非常感谢作者的工作！终于有可以自行配置的OpenWrt 插件可用了。
网上的类似于L大的那些固件太庞大，集成了太多用不上的东西…
从SS时代开始就喜欢在 OpenWrt 原版里自己安装配置。

kuoruan, 2020-03-03:
我一般写 luci 考虑的并不是易用性，而是灵活性，将所有配置项暴露给用户，用户可以按自己的喜好任意配置，同时也能过滤掉一些伸手党。
另外，不喜欢复杂配置的用户可以选择使用自定义配置文件。

kukudemajia, 2020-04-04:
建议可以设置一个模式，普通模式和高级模式，毕竟大部分人都对这么多参数感到陌生而无从下手。

kukudemajia, 2020-04-04:
弃疗，直接在终端配置算了

xueyichen, 2020-05-26:
借问下，18.06版本安装后，没有luci，opkg list也没找到，是因为？
系统为原版，多谢。

oemaix, 2020-08-28:
我认为现在的配置方式就挺好, 完整功能都可以配置.

lostz, 2020-10-12:
我也很想要一个精简版本

RayleiLi, 2020-10-12:
我觉得现在的UI很好呀，每个功能都有。只是不知道为什么我编译出来的在router界面看不到inbound和outbound😭

moralrebuild, 2020-10-21:
题主的意见非常有意义。SS的用户界面比较合理，复杂度上兼顾了普通用户。V2RAY在我看来一直有些鸡肋，开发者把复杂度整的很高，编译后footprint很大，结果不仅内存消耗大，CPU杀手，吞吐量很低；易用性上突破下线（我看网上太多关于V2RAY配置故障的问题），反而催生了很多伸手党。我基本上只使用SS的v2ray plugin。现实也证明即便SS技术陈旧，破网成功率不如V2RAY，凭借易用性、高性能、开源友好度三项优势，依然是科学上网界的最受欢迎工具。

ColasDAD, 2020-12-17:
> 题主的意见非常有意义。SS的用户界面比较合理，复杂度上兼顾了普通用户。V2RAY在我看来一直有些鸡肋，开发者把复杂度整的很高，编译后footprint很大，结果不仅内存消耗大，CPU杀手，吞吐量很低；易用性上突破下线（我看网上太多关于V2RAY配置故障的问题），反而催生了很多伸手党。我基本上只使用SS的v2ray plugin。现实也证明即便SS技术陈旧，破网成功率不如V2RAY，凭借易用性、高性能、开源友好度三项优势，依然是科学上网界的最受欢迎工具。

SS用上v2ray-plugin和V2RAY效果可以一样的啊，都能做到TLS。同样我最喜欢的还是SS，平台支持广泛，小巧精湛。

closed 2021-04-12 by kuoruan