kuoruan/openwrt-v2ray #55: SIGILL: illegal instruction 在R5，R4工作正常

wuhelong, 2020-03-09:
Please attach the output of following commands.

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.2'
DISTRIB_REVISION='r10947-65030d81f3'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.2 r10947-65030d81f3'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.2"
VERSION_ID="19.07.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r10947-65030d81f3"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.2 r10947-65030d81f3"
arch all 1
arch noarch 1
arch mipsel_24kc 10

v2ray -config /var/etc/v2ray/v2ray.main.json
V2Ray 4.22.1 (OpenWrt) R5 (go1.14 linux/mipsle)
A unified platform for anti-censorship.
main: failed to read config file: /var/etc/v2ray/v2ray.main.json > v2ray.com/core/main/json: failed to execute v2ctl to convert config file. > v2ray.com/core/common/platform/ctlcmd: failed to execute v2ctl: SIGILL: illegal instruction
PC=0x47a488 m=6 sigcode=128

goroutine 1 [running]:
runtime.asyncPreempt()
        runtime/preempt_mipsx.s:40 +0x7c fp=0x1ced6cc sp=0x1ced5d8 pc=0x47a488
github.com/golang/protobuf/proto.unmarshalUint32Value(0x24e89c1, 0x1, 0x3d1706, 0x2f23acc, 0x0, 0x24e89c0, 0x2, 0x3d1707, 0x0, 0x0)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:985 fp=0x1ced6d0 sp=0x1ced6d0 pc=0x579248
github.com/golang/protobuf/proto.(*unmarshalInfo).unmarshal(0x1c24d20, 0x2f23ac0, 0x24e89ba, 0x8, 0x3d170d, 0x1, 0x0)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:173 +0x800 fp=0x1ced7b8 sp=0x1ced6d0 pc=0x5732c4
github.com/golang/protobuf/proto.makeUnmarshalMessageSlicePtr.func1(0x24e89ba, 0x188a, 0x3d170d, 0x2d04818, 0x2, 0x24e89b8, 0x188c, 0x3d170f, 0x0, 0x0)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:1672 +0x17c fp=0x1ced808 sp=0x1ced7b8 pc=0x5915c4
github.com/golang/protobuf/proto.(*unmarshalInfo).unmarshal(0x1c24c60, 0x2d04810, 0x24e816c, 0x20d8, 0x3d1f5b, 0x2, 0x0)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:173 +0x800 fp=0x1ced8f0 sp=0x1ced808 pc=0x5732c4
github.com/golang/protobuf/proto.makeUnmarshalMessageSlicePtr.func1(0x24e816c, 0x3d1f5b, 0x3d1f5b, 0x207e120, 0x2, 0x24e8169, 0x3d1f5e, 0x3d1f5e, 0x0, 0x0)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:1672 +0x17c fp=0x1ced940 sp=0x1ced8f0 pc=0x5915c4
github.com/golang/protobuf/proto.(*unmarshalInfo).unmarshal(0x1c24ba0, 0x207e120, 0x2380000, 0x53a0c7, 0x53a0c7, 0xd01104, 0x40ba1c)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:173 +0x800 fp=0x1ceda28 sp=0x1ced940 pc=0x5732c4
github.com/golang/protobuf/proto.(*InternalMessageInfo).Unmarshal(0x1374410, 0xe77e30, 0x207e120, 0x2380000, 0x53a0c7, 0x53a0c7, 0x53a001, 0x6fbef0b0)
        github.com/golang/protobuf@v1.3.2/proto/table_unmarshal.go:63 +0x78 fp=0x1ceda58 sp=0x1ceda28 pc=0x572804
v2ray.com/core/app/router.(*GeoIPList).XXX_Unmarshal(0x207e120, 0x2380000, 0x53a0c7, 0x53a0c7, 0x207e120, 0x1c2aa01)
        v2ray.com/core/app/router/config.pb.go:351 +0x6c fp=0x1ceda7c sp=0x1ceda58 pc=0x98b190
github.com/golang/protobuf/proto.Unmarshal(0x2380000, 0x53a0c7, 0x53a0c7, 0xe77e30, 0x207e120, 0x0, 0x0)
        github.com/golang/protobuf@v1.3.2/proto/decode.go:337 +0x1b0 fp=0x1cedab0 sp=0x1ceda7c pc=0x54a2a8
v2ray.com/core/infra/conf.loadIP(0xd37751, 0x9, 0x1d15690, 0x7, 0x0, 0x526ad0, 0x0, 0x55c344, 0x0)
        v2ray.com/core/infra/conf/router.go:155 +0x228 fp=0x1cedb04 sp=0x1cedab0 pc=0xb75e9c
v2ray.com/core/infra/conf.loadGeoIP(...)
        v2ray.com/core/infra/conf/router.go:146
v2ray.com/core/infra/conf.toCidrList(0x1ceba00, 0x2, 0x4, 0xc15760, 0x1c633c0, 0x0, 0x0, 0x1d1c070)
        v2ray.com/core/infra/conf/router.go:306 +0x128 fp=0x1cedbd0 sp=0x1cedb04 pc=0xb778d0
v2ray.com/core/infra/conf.parseFieldRule(0x1d1c070, 0x70, 0x70, 0xc157a0, 0x1ceb880, 0x0)
        v2ray.com/core/infra/conf/router.go:398 +0x6c8 fp=0x1cedc58 sp=0x1cedbd0 pc=0xb78cc0
v2ray.com/core/infra/conf.ParseRule(0x1d1c070, 0x70, 0x70, 0x0, 0x77e34088, 0x0)
        v2ray.com/core/infra/conf/router.go:453 +0x588 fp=0x1cedcb4 sp=0x1cedc58 pc=0xb7979c
v2ray.com/core/infra/conf.(*RouterConfig).Build(0x1ceb780, 0x1c0d344, 0x3, 0x3)
        v2ray.com/core/infra/conf/router.go:75 +0x540 fp=0x1cedd10 sp=0x1cedcb4 pc=0xb75148
v2ray.com/core/infra/conf.(*Config).Build(0x1c250e0, 0xc6d000, 0x1c250e0, 0x0)
        v2ray.com/core/infra/conf/v2ray.go:362 +0x147c fp=0x1cede9c sp=0x1cedd10 pc=0xb85e10
v2ray.com/core/infra/conf/serial.LoadJSONConfig(0xe6ef30, 0x1c0e0e0, 0x1c80ea0, 0xbc2f88, 0xd348a7)
        v2ray.com/core/infra/conf/serial/loader.go:65 +0x618 fp=0x1cedeec sp=0x1cede9c pc=0xb8c47c
v2ray.com/core/infra/conf/command.(*ConfigCommand).Execute(0x1384060, 0x1c0c050, 0x0, 0x0, 0x1c80f20, 0x1)
        v2ray.com/core/infra/conf/command/command.go:30 +0x4c fp=0x1cedf30 sp=0x1cedeec pc=0xbc2ab4
main.main()
        v2ray.com/core/infra/control/main/main.go:32 +0x108 fp=0x1cedfc0 sp=0x1cedf30 pc=0xbc3008
runtime.main()
        runtime/proc.go:203 +0x284 fp=0x1cedfec sp=0x1cedfc0 pc=0x44248c
runtime.goexit()
        runtime/asm_mipsx.s:651 +0x4 fp=0x1cedfec sp=0x1cedfec pc=0x479d08

r0   0x0        r1   0x100
r2   0x0        r3   0x2f23ac0
r4   0x24e89c1  r5   0x0
r6   0x1        r7   0x3d1706
r8   0x0        r9   0x0
r10  0x10       r11  0x0
r12  0x18       r13  0x1c25170
r14  0x1c25184  r15  0x0
r16  0x2        r17  0x0
r18  0xf1       r19  0x8
r20  0x2f23ac0  r21  0x2ef7000
r22  0xd6bd9c   r23  0x6
r24  0x6fe26000 r25  0x6881a20
r26  0x812      r27  0x0
r28  0x137aa30  r29  0x1ced5d8
r30  0x1c000e0  r31  0x579248
pc   0x47a488   link 0x579248
lo   0x0        hi   0x0
 > exit status 2

kuoruan, 2020-03-10:
看样子是 v2ray 正常但 v2ctl 无法运行

wuhelong, 2020-03-10:
嗯，如果需要测试我可以帮忙

kuoruan, 2020-03-11:
https://openwrt.kuoruan.net/packages/snapshots/mipsel_24kc/v2ray-core_4.22.1-5_mipsel_24kc.ipk

试试这个文件

closed 2020-03-12 by kuoruan

wuhelong, 2020-03-16:
谢谢，它工作正常

commit bbfba38c (2020-03-12): chore: bump to v4.22.1-6

fix #55, #56