kuoruan/openwrt-v2ray #57: GFWList没有作用

tanghaojie, 2020-03-18:
Please attach the output of following commands.

```
cat /etc/*release

DISTRIB_ID='Openwrt Koolshare mod'
DISTRIB_RELEASE='V2.31'
DISTRIB_REVISION='r10822-50aa0525d1'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='Openwrt Koolshare mod V2.31 r10822-50aa0525d1'
DISTRIB_TAINTS='no-all busybox'
NAME="Openwrt Koolshare mod"
VERSION="V2.31"
ID="openwrt_koolshare_mod"
ID_LIKE="lede openwrt"
PRETTY_NAME="Openwrt Koolshare mod V2.31"
VERSION_ID="v2.31"
HOME_URL="http://openwrt.org/"
BUG_URL="http://koolshare.cn"
SUPPORT_URL="http://koolshare.cn"
BUILD_ID="r10822-50aa0525d1"
LEDE_BOARD="x86/64"
LEDE_ARCH="x86_64"
LEDE_TAINTS="no-all busybox"
LEDE_DEVICE_MANUFACTURER="koolshare"
LEDE_DEVICE_MANUFACTURER_URL="http://koolshare.cn"
LEDE_DEVICE_PRODUCT="fw867"
LEDE_DEVICE_REVISION="V2"
LEDE_RELEASE="Openwrt Koolshare mod V2.31 r10822-50aa0525d1"

opkg print-architecture

arch all 1
arch noarch 1
arch x86_64 10
```
透明代理，代理模式，设置为GFWList以后，整个外面没网的，测试Google和油管
[image]
我这里测试：
默认：全局代理
直连CN：CN白名单
代理CN：不知道？求解释
代理GFWList：应该是没有任何代理效果
[image]
两个马赛克都是我真实的ip，不是代理的ip

kuoruan, 2020-03-18:
防火墙规则借个图看看

tanghaojie, 2020-03-19:
> 防火墙规则借个图看看

[image]
[image]
[image]

kuoruan, 2020-03-19:
v2ray 那两条规则是你自己加的吗？
我的意思是开启代理时的防火墙规则。
也把配置文件贴一下

tanghaojie, 2020-03-19:
> v2ray 那两条规则是你自己加的吗？
> 我的意思是开启代理时的防火墙规则。
> 也把配置文件贴一下

噢，就是这个原因，我看教程说的要手动去防火墙里面加，原来不用加的啊。。。
谢谢大佬

tanghaojie, 2020-03-19:
原谅我又要打开了。。。测试的时候我忘记关本机代理了，其实是不行的

tanghaojie, 2020-03-19:
下面是配置：
[image]
[image]
[image]
[image]

接口：
[image]

防火墙v2ray的两条规则我删除了

kuoruan, 2020-03-19:
```
ipset list v2ray_dst_proxy_v4
```

tanghaojie, 2020-03-20:
> ipset list v2ray_dst_proxy_v4

[image]
一堆timeout，没截完

kuoruan, 2020-03-20:
这是正常的输出

把你的 GFWList 代理模式打开，然后贴上防火墙规则列表和关于里的配置文件输出。

tanghaojie, 2020-03-20:
> 这是正常的输出
> 
> 把你的 GFWList 代理模式打开，然后贴上防火墙规则列表和关于里的配置文件输出。

[image]
[image]
[image]
之前的截图都是打开了GFWList模式的

kuoruan, 2020-03-20:
```
iptables -t nat -nL
iptables -t mangle -nL
```

tanghaojie, 2020-03-20:
> ```
> iptables -t nat -nL
> iptables -t mangle -nL
> ```

>```
>iptables -t nat -nL
>````
[image]
[image]

>```
>iptables -t mangle -nL
>````
[image]

kuoruan, 2020-03-20:
配置看着没问题，看看v2ray的错误日志有没有什么发现

tanghaojie, 2020-03-20:
```
cat /var/log/v2ray-error.log
```
[image]

kuoruan, 2020-03-20:
把 direct 勾上。。。

kuoruan, 2020-03-20:
然后是 block 和 dns_out 都勾上

kuoruan, 2020-03-20:
你现在只有一个代理的出站，没有直连的出站和 dns 的出站

closed 2020-03-20 by kuoruan

AlexYe233, 2020-03-20:
kuoruan大佬能不能写个测试命令集合，让我们这些小白自行搞清楚问题在哪里
我现在初步解决了dns阻断与解析速度的问题，出站链接没有问题，我怀疑是在防火墙规则的域名没有顺利解析成ip
谢谢kuoruan大佬的付出，学习了

kuoruan, 2020-03-20:
ipset 列表中的 v2ray_dst_proxy_v4 有内容说明就解析正常的。

tanghaojie, 2020-03-20:
[image]
勾上了，现在日志里面没有错误了，但还是不行
[image]
这里面没内容了

kuoruan, 2020-03-20:
把 114.114.114.114 换为其他的国内 dns

具体的 v2ray 使用可以去 v2ray 论坛讨论

AlexYe233, 2020-03-20:
v2ray_dst_proxy_v4只有额外的代理列表怎么办呢，在几乎所有gfw的dns请求都被污染的环境下
v2ray开启了doh，能提供不污染的dns请求入口

kuoruan, 2020-03-20:
这个 ipset 列表会自动填充。
具体效果是，当 ```/etc/dnsmasq.d/v2ray``` 文件中的域名解析成功后会自动将对应的 IP 地址添加到此文件中

你 v2ray 中配置的 dns 已经设置了分流，不太清楚为什么没有正常解析出结果，可以尝试手动设置一下 代理列表使用的 dns 地址

AlexYe233, 2020-03-20:
这是我的v2配置(出站连接有删减)
"dns": {
		"servers": [
			"8.8.8.8",
			{
				"address": "https+local://",
				"port": 443,
				"domains": [
					"geosite:github"
				]
			},
			{
				"address": "localhost",
				"port": 53,
				"domains": [
					"geosite:cn"
				]
			}
		]
	},
	"routing": {
		"domainStrategy": "AsIs",
		"rules": [
			{
				"type": "field",
				"ip": [
					"192.168.1.1"
				],
				"port": "53",
				"inboundTag": [
					"transparent",
					"dns_in"
				],
				"outboundTag": "dns_out"
			},
			{
				"type": "field",
				"ip": [
					"geoip:private",
					"geoip:cn"
				],
				"outboundTag": "direct"
			},
			{
				"type": "field",
				"domain": [
					"geosite:cn"
				],
				"inboundTag": [
					"socks_proxy"
				],
				"outboundTag": "direct"
			}
		]
	},
	"inbounds": [
		{
			"listen": "127.0.0.1",
			"port": 1080,
			"protocol": "socks",
			"settings": {
				"auth": "noauth",
				"udp": true,
				"ip": "127.0.0.1"
			},
			"streamSettings": {
				"sockopt": {
					
				}
			},
			"tag": "socks_proxy",
			"sniffing": {
				"enabled": false
			}
		},
		{
			"port": 1081,
			"protocol": "dokodemo-door",
			"settings": {
				"followRedirect": true,
				"network": "tcp,udp"
			},
			"streamSettings": {
				"sockopt": {
					"tproxy": "tproxy"
				}
			},
			"tag": "transparent",
			"sniffing": {
				"enabled": false
			}
		}
	],
	"outbounds": [
		{
			"protocol": "vmess",
			"tag": "proxy",
			"mux": {
				"enabled": true,
				"concurrency": 8
			}
		},
		{
			"protocol": "freedom",
			"settings": {
				
			},
			"streamSettings": {
				"sockopt": {
					"mark": 255
				}
			},
			"tag": "direct"
		},
		{
			"protocol": "dns",
			"settings": {
				
			},
			"streamSettings": {
				"sockopt": {
					"mark": 255
				}
			},
			"tag": "dns_out"
		}
	]
}
这个是我的透明代理配置
[image]
ipset list v2ray_dst_proxy_v4只有8.8.8.8
域网设备向192.168.1.1发出的dns请求能得到正确结果，但是路由器向192.168.1.1发出的dns请求不能得到正确结果
麻烦大佬帮我看下哪里设置有问题

kuoruan, 2020-03-20:
路由里边填的 inboundTag 和 outboundTag 一定要和 出站和入站里边的 tag 对应，
你在路由里边加了个 dns_in 但是你并没有对应的 inbound 啊，我这里不处理 v2ray 的具体使用方法。你可以去看文档

kuoruan, 2020-03-20:
还有就是看看你客户端的 dns 配置，要设置为路由器的 ip 地址，不然 ipset 不起作用的

AlexYe233, 2020-03-20:
没有，那个dns_in不用的
现在就是搞不懂为什么在路由上nslookup www.google.com 192.168.1.1不行 但在连接路由的电脑和手机上可以

AlexYe233, 2020-03-20:
感觉好像在路由上能实现了的话v2ray_dst_proxy_v4就能获得正确地址了，
想请教一下我没有/etc/dnsmasq.d/v2ray这个文件（直接kuoruan.net源安装）
/etc/dnsmasq.d/v2ray 文件中的域名解析至v2ray_dst_proxy_v4是由哪个来实现的,这个过程dns请求会被转发至v2透明代理端口吗

kuoruan, 2020-03-20:
mangle 中的规则会把局域网内过来的 53 端口的 UDP 流量转发给 v2ray
理论上路由器自己的 53 端口的 UDP 流量也会被转发给 v2ray 处理，只是不知道为什么你没生效。

AlexYe233, 2020-03-20:
麻烦dalao得空帮我看一下nat表和mangle表设置有没有什么问题
iptables -t mangle -nL:
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
V2RAY      all  --  0.0.0.0/0            0.0.0.0/0           
V2RAY      all  --  0.0.0.0/0            0.0.0.0/0            mark match 0x1
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
TCPMSS     tcp  --  0.0.0.0/0            0.0.0.0/0            tcp flags:0x06/0x02 /* !fw3: Zone wan MTU fixing */ TCPMSS clamp to PMTU
Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
V2RAY_MASK  all  --  0.0.0.0/0            0.0.0.0/0           
Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
Chain V2RAY (2 references)
target     prot opt source               destination         
RETURN     all  --  0.0.0.0/0            0.0.0.0/0            mark match 0xff
TPROXY     udp  --  0.0.0.0/0            192.168.1.1          udp dpt:53 TPROXY redirect 0.0.0.0:1081 mark 0x1/0x1
RETURN     all  --  0.0.0.0/0            0.0.0.0/0            match-set v2ray_dst_direct_v4 dst
TPROXY     tcp  --  0.0.0.0/0            0.0.0.0/0            match-set v2ray_dst_proxy_v4 dst TPROXY redirect 0.0.0.0:1081 mark 0x1/0x1
TPROXY     udp  --  0.0.0.0/0            0.0.0.0/0            TPROXY redirect 0.0.0.0:1081 mark 0x1/0x1
Chain V2RAY_MASK (1 references)
target     prot opt source               destination         
RETURN     all  --  0.0.0.0/0            0.0.0.0/0            mark match 0xff
RETURN     all  --  0.0.0.0/0            0.0.0.0/0            match-set v2ray_dst_direct_v4 dst
MARK       tcp  --  0.0.0.0/0            0.0.0.0/0            match-set v2ray_dst_proxy_v4 dst MARK set 0x1
MARK       udp  --  0.0.0.0/0            0.0.0.0/0            MARK set 0x1

iptables -t nat -nL:
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
prerouting_rule  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom prerouting rule chain */
zone_lan_prerouting  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
zone_wan_prerouting  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
postrouting_rule  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom postrouting rule chain */
zone_lan_postrouting  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
zone_wan_postrouting  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
Chain postrouting_lan_rule (1 references)
target     prot opt source               destination         
Chain postrouting_rule (1 references)
target     prot opt source               destination         
Chain postrouting_wan_rule (1 references)
target     prot opt source               destination        
Chain prerouting_lan_rule (1 references)
target     prot opt source               destination         
Chain prerouting_rule (1 references)
target     prot opt source               destination         
Chain prerouting_wan_rule (1 references)
target     prot opt source               destination         
Chain zone_lan_postrouting (1 references)
target     prot opt source               destination         
postrouting_lan_rule  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom lan postrouting rule chain */
Chain zone_lan_prerouting (1 references)
target     prot opt source               destination         
prerouting_lan_rule  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom lan prerouting rule chain */
Chain zone_wan_postrouting (1 references)
target     prot opt source               destination         
postrouting_wan_rule  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom wan postrouting rule chain */
MASQUERADE  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3 */
Chain zone_wan_prerouting (1 references)
target     prot opt source               destination         
prerouting_wan_rule  all  --  0.0.0.0/0            0.0.0.0/0            /* !fw3: Custom wan prerouting rule chain */

kuoruan, 2020-03-20:
正常情况下 nat 表应该有 v2ray luci 插入的一个 V2RAY 链，你这个不见了

可以看上面的截图。

AlexYe233, 2020-03-21:
大佬，怎么手动添加这些记录进nat表呢
缺少的记录
Chain PREROUTING (policy ACCEPT)
V2RAY		tcp --	0.0.0.0/0	0.0.0.0/0
Chain OUTPUT (policy ACCEPT)
V2RAY		tcp --	0.0.0.0/0	0.0.0.0/0
Chain V2RAY
RETURN		tcp	--	0.0.0.0/0	0.0.0.0/0	mark match 0xff
RETURN		tcp	--	0.0.0.0/0	0.0.0.0/0	match-set v2ray_dst_direct_v4 dst
REDIRECT	tcp	--	0.0.0.0/0	0.0.0.0/0	match-set v2ray_dst_proxy_v4 dst redir ports 1081

kuoruan, 2020-03-21:
@AlexYe233 你的防火墙规则是正常的，TProxy 下不向 nat 表写规则。

AlexYe233, 2020-03-21:
欸，搞不懂为什么在路由上nslookup www.google.com 192.168.1.1不行 但在连接路由的电脑和手机上就可以

AlexYe233, 2020-03-22:
大佬您好，
防火墙mangle表的调整后，现在路由上的dns服务器192.168.1.1已经能解析所有网站出正确的结果了
我的路由上没有/etc/dnsmasq.d/v2ray，只找到了/tmp/dnsmasq.d/v2ray
其中内容为
server=/030buy.com/192.168.1.1
ipset=/030buy.com/v2ray_dst_proxy_v4,v2ray_dst_proxy_v6
server=/0rz.tw/192.168.1.1
ipset=/0rz.tw/v2ray_dst_proxy_v4,v2ray_dst_proxy_v6
……
但是/tmp/dnsmasq.d/v2ray仍不能填充至v2ray_dst_proxy_v4列表
请问是您知道原因吗，或者您推测问题大致在哪里

kukudemajia, 2020-04-05:
这个问题我也遇到，也对比了楼主的相关情况都差不多
我的解决办法就是卸载重装
但是gfw list还是不稳定，只要你修改了某个参数就很有可能gfw list就没法翻了，就算你恢复原来的参数，删除/tmp/dnsmasq.d/v2ray，也是一样的结果
有时候选择直连cn再切换回来又可以了
现在就是怎么也不行，这个确实有问题
默认（直连CN）一直没有问题，就是费流量，国外的IP都会默认走v2ray