kuoruan/openwrt-v2ray #60: GFWlist的问题

kukudemajia, 2020-04-05:
Please attach the output of following commands.

```
cat /etc/*release

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.2'
DISTRIB_REVISION='r10947-65030d81f3'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 19.07.2 r10947-65030d81f3'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.2"
VERSION_ID="19.07.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r10947-65030d81f3"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.2 r10947-65030d81f3"

opkg print-architecture

arch all 1
arch noarch 1
arch x86_64 10

```
大概说一下：
在发帖之前也仔细看了一下其他的，基本上情况类似。
用直连CN（默认）一直很稳定，youtube速度也正常，大概在30000kbps
但是用GFWlist速度就非常慢，youtube速度大概在2000kbps，甚至更低
GFWlist模式下只要修改了其中一个参数，即使后来恢复了原有配置，也有大概率无法通过v2ray翻出去，删除/etc/dnsmasq.d/v2ray，重启都试过也不行，偶尔通过切换模式就恢复正常。
直连CN虽好，但是费流量，毕竟国外的IP都走流量
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kuoruan, 2020-04-06:
为代理列表和直连列表都设置上 DNS 服务器，然后添加路由，让对应的 DNS IP 走直连或者代理

kukudemajia, 2020-04-06:
> 为代理列表和直连列表都设置上 DNS 服务器，然后添加路由，让对应的 DNS IP 走直连或者代理

改了DNS 服务器里面的服务器好像正常了，代理列表和直连列表都没有设置上 DNS 服务器

closed 2020-04-06 by kukudemajia

nico-johnny, 2020-11-14:
> > 为代理列表和直连列表都设置上 DNS 服务器，然后添加路由，让对应的 DNS IP 走直连或者代理
> 
> 改了DNS 服务器里面的服务器好像正常了，代理列表和直连列表都没有设置上 DNS 服务器

请问是怎么改的？