kuoruan/openwrt-v2ray #61: 内存占用异常

Flying-Tiantian, 2020-04-06:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.2'
DISTRIB_REVISION='r10947-65030d81f3'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 19.07.2 r10947-65030d81f3'
DISTRIB_TAINTS=''
```
OpenWRT跑在PVE虚拟机里，一共分配了1GB内存，使用v2ray-core_4.23.1-1_x86_64.ipk
，v2ray服务的内存占用达到704MB:

  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1862     1 root     S     704m  71%   0% /usr/bin/v2ray --config=/var/etc/v2ray/v2ray.main.json

VSZ不代表实际使用内存，但是这预分配内存也太多了吧
进而导致另一个问题，通过luci-app-v2ray限制内存使用量为50%，会导致服务无法启动，而且没有任何报错（也可能是我没找到。。。）

Flying-Tiantian, 2020-04-06:
"
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: fatal error: failed to reserve page summary memory
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime stack:
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.throw(0xd2069f, 0x25)
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/panic.go:1114 +0x72 fp=0x7fff9b1482d8 sp=0x7fff9b1482a8 pc=0x4356c2
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.(*pageAlloc).sysInit(0x13db888)
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/mpagealloc_64bit.go:80 +0x17e fp=0x7fff9b148368 sp=0x7fff9b1482d8 pc=0x42c10e
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.(*pageAlloc).init(0x13db888, 0x13db880, 0x13f03b8)
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/mpagealloc.go:297 +0x75 fp=0x7fff9b148390 sp=0x7fff9b148368 pc=0x429d65
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.(*mheap).init(0x13db880)
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/mheap.go:694 +0x274 fp=0x7fff9b1483b8 sp=0x7fff9b148390 pc=0x426f84
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.mallocinit()
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/malloc.go:470 +0xff fp=0x7fff9b1483e8 sp=0x7fff9b1483b8 pc=0x40d34f
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.schedinit()
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/proc.go:545 +0x60 fp=0x7fff9b148440 sp=0x7fff9b1483e8 pc=0x438de0
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]: runtime.rt0_go(0x7fff9b1484a8, 0x2, 0x7fff9b1484a8, 0x0, 0x7f3959a2177f, 0x7fff9b1484a8, 0x7fff9b1484a0, 0x0, 0x108, 0x7f3959a73d68, ...)
Tue Apr  7 00:18:46 2020 daemon.err v2ray[6267]:        runtime/asm_amd64.s:214 +0x125 fp=0x7fff9b148448 sp=0x7fff9b148440 pc=0x462f65
"

附上log

kuoruan, 2020-04-07:
用官方的编译试试吧，可能是 1.14.1 的问题

Flying-Tiantian, 2020-04-07:
用OpenWRT的SDK编译还是一样的问题，现在在用[v2ray官方](https://github.com/v2ray/v2ray-core/releases)的可执行文件，VSZ下降到117MB了。

kuoruan, 2020-04-07:
https://github.com/kuoruan/openwrt-v2ray/releases/tag/v4.23.1-2

试试 1.13.8

Flying-Tiantian, 2020-04-07:
OK！正常了，老哥威武。

closed 2020-04-07 by kuoruan