kuoruan/openwrt-v2ray #67: 刷的最新的固件，报错Insufficient permissions to read UCI configuration.

xinwf, 2020-04-27:
Please attach the output of following commands.

```
cat /etc/*release
opkg print-architecture
```
```
DISTRIB_ID='X-WRT'
DISTRIB_RELEASE='8.0_b202004241024'
DISTRIB_REVISION='r13146-fc6d682123'
DISTRIB_CODENAME='focal'
DISTRIB_TARGET='ramips/mt7620'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='X-WRT 8.0_b202004241024 Focal'
DISTRIB_TAINTS='busybox'
NAME="X-WRT"
VERSION="8.0_b202004241024"
ID="x-wrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="X-WRT 8.0_b202004241024"
VERSION_ID="8.0_b202004241024"
HOME_URL="https://x-wrt.com/"
BUG_URL="https://x-wrt.com/"
SUPPORT_URL="https://x-wrt.com/"
BUILD_ID="r13146-fc6d682123"
OPENWRT_BOARD="ramips/mt7620"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="X-WRT-TEAM"
OPENWRT_DEVICE_MANUFACTURER_URL="https://x-wrt.com/"
OPENWRT_DEVICE_PRODUCT="X-Router"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="X-WRT 8.0_b202004241024 Focal"
```

```
arch all 1
arch noarch 1
arch mipsel_24kc 10
```

[image]

xinwf, 2020-04-28:
这个问题已解决, 根据 https://github.com/LGA1150/openwrt/commit/c822400a5da3ac6b093ce0dd50efccc626759d27 这个链接在路由器上创建这个文件/usr/share/rpcd/acl.d/luci-app-vlmcsd.json, 然后打开文件read 和write 加上v2ray即可, 这个问题解决后,我的v2ray服务无法启动, 提示无法load config文件,  v2ray 4.18 failded to load config

kuoruan, 2020-04-29:
新版本的 LuCI2 使用了全新的架构，并引入了权限控制，我会为新版重写一个 LuCI，完成时间未定。

closed 2020-04-29 by kuoruan