kuoruan/openwrt-v2ray #68: /var/etc/v2ray/v2ray.main.json文件不存在

xinwf, 2020-04-28:
Please attach the output of following commands.

```
cat /etc/*release
opkg print-architecture
```
```
DISTRIB_ID='X-WRT'
DISTRIB_RELEASE='8.0_b202004241024'
DISTRIB_REVISION='r13146-fc6d682123'
DISTRIB_CODENAME='focal'
DISTRIB_TARGET='ramips/mt7620'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='X-WRT 8.0_b202004241024 Focal'
DISTRIB_TAINTS='busybox'
NAME="X-WRT"
VERSION="8.0_b202004241024"
ID="x-wrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="X-WRT 8.0_b202004241024"
VERSION_ID="8.0_b202004241024"
HOME_URL="https://x-wrt.com/"
BUG_URL="https://x-wrt.com/"
SUPPORT_URL="https://x-wrt.com/"
BUILD_ID="r13146-fc6d682123"
OPENWRT_BOARD="ramips/mt7620"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="X-WRT-TEAM"
OPENWRT_DEVICE_MANUFACTURER_URL="https://x-wrt.com/"
OPENWRT_DEVICE_PRODUCT="X-Router"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="X-WRT 8.0_b202004241024 Focal"
```
```
arch all 1
arch noarch 1
arch mipsel_24kc 10
```
权限的问题解决后, 可以看的v2ray的服务界面, 在关于里面看到当前的配置文件在`/var/etc/v2ray/v2ray.main.json`, 这个文件确实不存在, 而且`/var`是映射的`/tmp`文件夹,这意味着路由器每次开机重启都会重新创建这个文件, 可是却没有, 这地方是忘了吧, 我把自己的配置文件弄过来,可以正常显示, 但是v2ray服务未运行

xinwf, 2020-04-28:
v2ray 服务尝试了很多次,都无法启动
[image]

kuoruan, 2020-04-29:
这个文件本身就是自动生成的，修改它也没有意义。

closed 2020-04-29 by kuoruan

xinwf, 2020-04-29:
> 这个文件本身就是自动生成的，修改它也没有意义。

问题在于它并没有自动生成，连v2ray 文件夹都没有