kuoruan/openwrt-v2ray #71: 能否增加一个软协处理器的编译选项

nobk, 2020-05-03:
我的路由器硬件是rampis/MT7621，这个CPU不带FPU协处理器，所以v2ray的浮点计算需要软件模拟。
在v2ray-core官网提供的v2ray-linux-mipsel-4.23.1zip里面，执行文件分成了
v2ray和v2ray_softfloat，以及v2ctl和v2ctl_softfloat两类，希望这边也能提供类似的选项，让生成的
v2ray和v2ctl能支持MT7621。

kuoruan, 2020-05-03:
OpenWrt 在编译过程中会自动为不支持 FPU 的设备添加 softfloat，不用特意处理。

closed 2020-05-03 by kuoruan

nobk, 2020-05-03:
关掉了UPX打包选项后重新生成了一个ipk包，的确如你所说已经是softfloat编译的。

nobk, 2020-05-03:
重新打包发现firmware大了整整2M，拆包仔细比较了一下，生成的v2ctl比官网的大太多了，v2ray倒是因为我关掉shadowsocks协议支持要小一点。
zsh#root@OpenWrt /tmp/v2/usr/bin
ll
-rwxr-xr-x    1 root     root       15.4M May  1 23:19 v2ctl
-rwxr-xr-x    1 root     root       15.1M May  1 23:19 v2ray
zsh#root@OpenWrt /tmp/v2/usr/bin
ll /usr/bin/v2*
-rwxr-xr-x    1 root     root       10.3M Apr 28 06:48 /usr/bin/v2ctl
-rwxr-xr-x    1 root     root       15.3M Apr 29 06:10 /usr/bin/v2ray

nobk, 2020-05-03:
最后还是用上不生成v2ctl的选项了，刷到主路由跑两天试试看。