kuoruan/openwrt-v2ray #72: 請問位置不夠的話要怎騰空位置出來?

aidanchan0101, 2020-05-12:
* verify_pkg_installable: Only have 3052kb available on filesystem /overlay, pkg v2ray-core needs 12503
 * opkg_install_cmd: Cannot install package v2ray-core.

kuoruan, 2020-05-13:
请手动安装 minimal

aidanchan0101, 2020-05-15:
然後就出現這個問題
pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core found, but incompatible with the architectures configured

我看到有人有相同問題，請問要怎解決呢?

opkg print-architecture是：
arch all 1
arch noarch 1
arch mipsel_24kc 1

NAME="OpenWrt"
VERSION="18.06.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 18.06.4"
VERSION_ID="18.06.4"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r7808-ef686b7292"
LEDE_BOARD="ramips/mt7621"
LEDE_ARCH="mipsel_24kc"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt 18.06.4 r7808-ef686b7292"

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06.4'
DISTRIB_REVISION='r7808-ef686b7292'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 18.06.4 r7808-ef686b7292'
DISTRIB_TAINTS=''

closed 2020-05-17 by aidanchan0101