kuoruan/openwrt-v2ray #77: 请问如何查性能瓶颈

lvkai, 2020-05-19:
从电脑直连可以流畅看油管8K，通过软路由透明代理只能流畅看HD，看4K，8K很卡。

我看CPU和内存使用率都不算高。
Mem: 86236K used, 167484K free, 4088K shrd, 2196K buff, 32952K cached
CPU:  21% usr   3% sys   0% nic  70% idle   0% io   0% irq   4% sirq
Load average: 2.85 1.03 0.58 3/80 1584

Please attach the output of following commands.

```
root@OpenWrt:~# cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.2'
DISTRIB_REVISION='r10947-65030d81f3'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.2 r10947-65030d81f3'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.2"
VERSION_ID="19.07.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r10947-65030d81f3"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.2 r10947-65030d81f3"
root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10

```

lvkai, 2020-05-19:
v2ray是不是只跑在一个虚拟核上并且跑满了？

kuoruan, 2020-05-19:
1. 检查日志是否有报错；
2. 到官方仓库提 issue。

closed 2020-05-19 by kuoruan