kuoruan/openwrt-v2ray #78: Cannot satisfy the following dependencies for v2ray-core:  ca-certificates

tangtao-xp1, 2020-05-24:
device: Xiao Mi Router 3
os version: X-WRT 8.0_b202005150236, Focal (a chinese fork of open-wrt)

when i try to manual install, it failed as below:
---------------------------------------------------------------------------------------------
root@X-WRT:~/tmp# opkg install minimal.v2ray-core_4.23.1-5_mipsel_24kc.ipk
Installing v2ray-core (4.23.1-5) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for v2ray-core:
 *      ca-certificates
 * opkg_install_cmd: Cannot install package v2ray-core.
----------------------------------------------------------------------------------------------
pls help, thank you!

kuoruan, 2020-05-25:
Did you do `opkg update` first?

tangtao-xp1, 2020-05-25:
> Did you do `opkg update` first?

thanks for your reply!

yes, and i tried to **opkg update and opkg install ca-certificates**, and I was told there is no dependencies named 'ca-certificates'

kuoruan, 2020-05-25:
Try this: https://downloads.openwrt.org/releases/packages-19.07/mipsel_24kc/base/ca-certificates_20190110-2_all.ipk

closed 2020-05-25 by kuoruan