kuoruan/openwrt-v2ray #79: 请教我该用哪个安装包？

wyjtm, 2020-05-25:
主机名	RT-AC58U
型号	ASUS RT-AC58U
架构	ARMv7 Processor rev 5 (v7l)
固件版本	OpenWrt 19.07.2 r10947-65030d81f3 / LuCI openwrt-19.07 branch git-20.057.55219-13dd17f
内核版本	4.14.171

上面是硬件
试了v2ray-core_4.23.1-5_arm_cortex-a7_neon-vfpv4.ipk提示

Executing package manager
Installing v2ray-core (4.23.1-5) to root...
Errors
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for v2ray-core:
 * 	ca-certificates
 * opkg_install_cmd: Cannot install package v2ray-core.
The opkg install command failed with code 255.

试了其他的arm的包也都是这个错误
我该用哪个安装包？

kuoruan, 2020-05-25:
`opkg update && opkg install ca-certificates`

closed 2020-05-25 by kuoruan

wyjtm, 2020-05-25:
> `opkg update && opkg install ca-certificates`
我一个个下包给装上了 去装ui的时候提示

xecuting package manager
Installing luci-app-v2ray (1.5.6) to root...
Installing kmod-ipt-tproxy (4.14.171-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ipq40xx/generic/kmods/4.14.171-1-fa00c1231ac7d7840ec6ffe62dcad926/kmod-ipt-tproxy_4.14.171-1_arm_cortex-a7_neon-vfpv4.ipk
Installing iptables-mod-tproxy (1.8.3-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.2/targets/ipq40xx/generic/packages/iptables-mod-tproxy_1.8.3-1_arm_cortex-a7_neon-vfpv4.ipk
Installing dnsmasq-full (2.80-16.1) to root...
Configuring kmod-ipt-tproxy.
Configuring iptables-mod-tproxy.
Errors
Collected errors:
 * opkg_download_pkg: Package dnsmasq-full is not available from any configured src.
 * opkg_install_pkg: Failed to download dnsmasq-full. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package luci-app-v2ray.

dnsmasq-full包也下了 装不上
卡到这一步进行不下去
@kuoruan

kuoruan, 2020-05-25:
`opkg remove dnsmasq && opkg install dnsmasq-full`

wyjtm, 2020-05-26:
> `opkg remove dnsmasq && opkg install dnsmasq-full`
折腾了半天装是装上了  折腾死我了
没找到订阅在哪
@kuoruan