kuoruan/openwrt-v2ray #82: process crash immediatly after launch

StoneMoe, 2020-06-01:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.2'
DISTRIB_REVISION='r10947-65030d81f3'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 19.07.2 r10947-65030d81f3'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.2"
VERSION_ID="19.07.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r10947-65030d81f3"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.2 r10947-65030d81f3"

arch all 1
arch noarch 1
arch x86_64 10
```

```
Mon Jun  1 20:10:07 2020 daemon.info v2ray: Policy disabled: main_policy
Mon Jun  1 20:10:07 2020 daemon.info v2ray: Reverse disabled: main_reverse
Mon Jun  1 20:10:08 2020 daemon.info v2ray: Setting transparent proxy on port: 12000
Mon Jun  1 20:10:08 2020 daemon.info v2ray: Transparent proxy mode: default
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: fatal error: failed to reserve page summary memory
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]:
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime stack:
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.throw(0xd2069f, 0x25)
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/panic.go:1116 +0x72 fp=0x7ffce4b146e8 sp=0x7ffce4b146b8 pc=0x435762
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.(*pageAlloc).sysInit(0x13dc8c8)
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/mpagealloc_64bit.go:80 +0x17e fp=0x7ffce4b14778 sp=0x7ffce4b146e8 pc=0x42c1ce
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.(*pageAlloc).init(0x13dc8c8, 0x13dc8c0, 0x13f13f8)
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/mpagealloc.go:297 +0x75 fp=0x7ffce4b147a0 sp=0x7ffce4b14778 pc=0x429e25
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.(*mheap).init(0x13dc8c0)
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/mheap.go:694 +0x274 fp=0x7ffce4b147c8 sp=0x7ffce4b147a0 pc=0x427044
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.mallocinit()
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/malloc.go:470 +0xff fp=0x7ffce4b147f8 sp=0x7ffce4b147c8 pc=0x40d34f
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.schedinit()
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/proc.go:545 +0x60 fp=0x7ffce4b14850 sp=0x7ffce4b147f8 pc=0x438e80
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: runtime.rt0_go(0x7ffce4b148b8, 0x2, 0x7ffce4b148b8, 0x0, 0x7f08d883677f, 0x7ffce4b148b8, 0x7ffce4b148b0, 0x0, 0x108, 0x7f08d8888d68, ...)
Mon Jun  1 20:10:08 2020 daemon.err v2ray[5299]: 	runtime/asm_amd64.s:214 +0x125 fp=0x7ffce4b14858 sp=0x7ffce4b14850 pc=0x4632b5
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: fatal error: failed to reserve page summary memory
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]:
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime stack:
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.throw(0xd2069f, 0x25)
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/panic.go:1116 +0x72 fp=0x7fff9dec7758 sp=0x7fff9dec7728 pc=0x435762
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.(*pageAlloc).sysInit(0x13dc8c8)
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/mpagealloc_64bit.go:80 +0x17e fp=0x7fff9dec77e8 sp=0x7fff9dec7758 pc=0x42c1ce
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.(*pageAlloc).init(0x13dc8c8, 0x13dc8c0, 0x13f13f8)
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/mpagealloc.go:297 +0x75 fp=0x7fff9dec7810 sp=0x7fff9dec77e8 pc=0x429e25
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.(*mheap).init(0x13dc8c0)
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/mheap.go:694 +0x274 fp=0x7fff9dec7838 sp=0x7fff9dec7810 pc=0x427044
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.mallocinit()
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/malloc.go:470 +0xff fp=0x7fff9dec7868 sp=0x7fff9dec7838 pc=0x40d34f
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.schedinit()
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/proc.go:545 +0x60 fp=0x7fff9dec78c0 sp=0x7fff9dec7868 pc=0x438e80
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: runtime.rt0_go(0x7fff9dec7928, 0x2, 0x7fff9dec7928, 0x0, 0x7fd1d945077f, 0x7fff9dec7928, 0x7fff9dec7920, 0x0, 0x108, 0x7fd1d94a2d68, ...)
Mon Jun  1 20:10:13 2020 daemon.err v2ray[5333]: 	runtime/asm_amd64.s:214 +0x125 fp=0x7fff9dec78c8 sp=0x7fff9dec78c0 pc=0x4632b5
Mon Jun  1 20:10:18 2020 daemon.info v2ray: Service disabled: main
Mon Jun  1 20:10:18 2020 daemon.info v2ray: No v2ray instance enabled, skip transparent proxy.
```

kuoruan, 2020-06-01:
If you are using LuCI, set `Memory percentage` to 0

StoneMoe, 2020-06-01:
It works :)

closed 2020-06-01 by StoneMoe