kuoruan/openwrt-v2ray #84: 安装完没有文件

zhangguanzhang, 2020-06-11:
Please attach the output of following commands.

```
root@OpenWrt:/usr/bin# cat /etc/*release
# PLEASE DO NOT EDIT THIS FILE
BOARD=aml-s9xxx
BOARD_NAME="AML-S9xxx"
BOARDFAMILY=aml-s9xxx
BUILD_REPOSITORY_URL=https://github.com/150balbes/Build-Armbian.git
BUILD_REPOSITORY_COMMIT=62b8c07-dirty
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=20.02.0-rc1.037
LINUXFAMILY=aml-s9xxx
BRANCH=current
ARCH=arm64
IMAGE_TYPE=nightly
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
# PLEASE DO NOT EDIT THIS FILE
BOARD=aml-s9xxx
BOARD_NAME="AML-S9xxx"
BOARDFAMILY=aml-s9xxx
BUILD_REPOSITORY_URL=https://github.com/150balbes/Build-Armbian.git
BUILD_REPOSITORY_COMMIT=62b8c07-dirty
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=20.02.0-rc1.037
LINUXFAMILY=aml-s9xxx
BRANCH=current
ARCH=arm64
IMAGE_TYPE=nightly
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_TARGET='armvirt/64'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_TAINTS='no-all'
DISTRIB_REVISION='R20.4.8(Compiled by gd772 on May 2, 2020)'
DISTRIB_DESCRIPTION='OpenWrt '
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r2499-01aeb5df"
OPENWRT_BOARD="armvirt/64"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r2499-01aeb5df"
root@OpenWrt:/usr/bin# opkg print-architecture
arch all 1
arch noarch 1
arch aarch64_generic 10
root@OpenWrt:/usr/bin# 
```

```
$ ps | grep v[2]
$ opkg files v2ray-core
Package v2ray-core (4.23.4-1) is installed on root and has the following files:
/usr/bin/v2ray
/usr/bin/geoip.dat
/usr/bin/geosite.dat
/usr/bin/v2ctl
$ opkg install v2ray-core
Package v2ray-core (4.23.4-1) installed in root is up to date.
$  ll /usr/bin/v2ray
ls: /usr/bin/v2ray: No such file or directory

```

kuoruan, 2020-06-11:
可能空间不足造成没装上，请卸载重试！

closed 2020-06-11 by kuoruan

zhangguanzhang, 2020-06-11:
# df -h
Filesystem                Size      Used Available Use% Mounted on
udev                    512.0K         0    512.0K   0% /dev
tmpfs                   174.3M     68.0K    174.3M   0% /run
/dev/mmcblk1p2          512.0M    369.9M     82.1M  82% /
tmpfs                   871.7M     19.2M    852.5M   2% /tmp
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk1p3            6.0G    317.7M      5.4G   5% /mnt/mmcblk1p3
/dev/mmcblk1p1          126.0M     75.9M     50.1M  60% /boot
cgroup                  871.7M         0    871.7M   0% /sys/fs/cgroup
/dev/mmcblk1p3            6.0G    317.7M      5.4G   5% /mnt/mmcblk1p3/docker
/dev/mmcblk1p3            6.0G    317.7M      5.4G   5% /mnt/mmcblk1p3/docker/btrfs