kuoruan/openwrt-v2ray #88: wfopen: /usr/bin/v2ray: Is a directory

Rangtian, 2020-06-25:
`

Please` attach the output of following commands.

```
cat /etc/*release

`# PLEASE DO NOT EDIT THIS FILE
BOARD=aml-s9xxx
BOARD_NAME="AML-S9xxx"
BOARDFAMILY=aml-s9xxx
BUILD_REPOSITORY_URL=https://github.com/150balbes/Build-Armbian.git
BUILD_REPOSITORY_COMMIT=62b8c07-dirty
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=20.02.0-rc1.037
LINUXFAMILY=aml-s9xxx
BRANCH=current
ARCH=arm64
IMAGE_TYPE=nightly
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
# PLEASE DO NOT EDIT THIS FILE
BOARD=aml-s9xxx
BOARD_NAME="AML-S9xxx"
BOARDFAMILY=aml-s9xxx
BUILD_REPOSITORY_URL=https://github.com/150balbes/Build-Armbian.git
BUILD_REPOSITORY_COMMIT=62b8c07-dirty
DISTRIBUTION_CODENAME=buster
DISTRIBUTION_STATUS=supported
VERSION=20.02.0-rc1.037
LINUXFAMILY=aml-s9xxx
BRANCH=current
ARCH=arm64
IMAGE_TYPE=nightly
BOARD_TYPE=conf
INITRD_ARCH=arm64
KERNEL_IMAGE_TYPE=Image
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_TARGET='armvirt/64'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_TAINTS='no-all'
DISTRIB_REVISION='R20.5.20'
DISTRIB_DESCRIPTION='OpenWrt '
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r2583-cad7fb2a"
OPENWRT_BOARD="armvirt/64"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r2583-cad7fb2a"`

opkg print-architecture

`arch all 1
arch noarch 1
arch aarch64_generic 10`

报错如下:

`root@OpenWrt:/etc/opkg# opkg upgrade v2ray-core
Installing v2ray-core (4.25.1-1) to root...
Downloading https://openwrt.kuoruan.net/packages/releases/aarch64_generic/v2ray-core_4.25.1-1_aarch64_generic.ipk
Configuring v2ray-core.
Collected errors:
 * wfopen: /usr/bin/v2ray: Is a directory.`

Rangtian, 2020-06-25:
我到/usr/bin目录下瞅了眼，v2ray确实是dir......

ztfot, 2020-06-27:
所以这个怎么解决？
我把原来的v2ray改为其他名字后安装，照样无法安装，还是老版本
我是手动安装

kuoruan, 2020-06-28:
重命名一下目录
`mv /usr/bin/v2ray /usr/bin/v2ray.bak`

closed 2021-04-12 by kuoruan