kuoruan/openwrt-v2ray #90: v2ray-core-mini mipsel_24kc无法正常运行

cyber386, 2020-06-26:
mipsel_24kc架构的v2ray-core-mini无法在mt7621下正常运行，试了好几个版本都不行，设备为:[Newifi D2](https://openwrt.org/toh/hwdata/d-team/d-team_newifi_d2)，表现为运行就卡住，无任何输出。

```
cat /etc/*release

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.3'
DISTRIB_REVISION='r11063-85e04e9f46'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.3 r11063-85e04e9f46'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.3"
VERSION_ID="19.07.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11063-85e04e9f46"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.3 r11063-85e04e9f46"

opkg print-architecture

arch all 1
arch noarch 1
arch mipsel_24kc 10
```

kuoruan, 2020-06-26:
先看看能不能输出版本号

cyber386, 2020-06-26:
> 先看看能不能输出版本号

不能，运行就卡住了，没有stdout 需Ctrl+C终止。感觉可能是upx压缩的问题。

cyber386, 2020-06-26:
还有是softfloat构建的吗？

kuoruan, 2020-06-26:
默认在 mipsel_24kc 是开启了 softfloat 的

saszsj, 2020-06-29:
我也碰到了 设备是 MT300N-V2
cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.3'
DISTRIB_REVISION='r11063-85e04e9f46'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.3 r11063-85e04e9f46'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.3"
VERSION_ID="19.07.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11063-85e04e9f46"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.3 r11063-85e04e9f46"

opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10

v2ray --version 也无法输出版本信息

kuoruan, 2020-06-29:
你们可以尝试自己编译一个 mini 但是关闭 upx 压缩试试

cyber386, 2020-06-29:
[这是](https://send.firefox.com/download/9b26946209466914/#Tppu5Yg4wnuZKQx3zvVGSw)未UPX的编译版本，我目前手边没有设备测试，有空的可以先试下。

cyber386, 2020-06-29:
[这是](https://send.firefox.com/download/f0b0d848d2bb1051/#IfVz2UQXPiI74lCchdsa_g)直接使用golang交叉编译并UPX压缩过的版本，也可以试下。

imapproxy, 2020-07-02:
可能是同样的问题，设备是极路由1s 5661A，系统19.07.3. 
安装顺利没有报错，但是浏览器进入路由器后台，看不到V2RAY；终端SSH进入路由器，v2ray --version同样无法输出版本号，卡住了。

cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.3'
DISTRIB_REVISION='r11063-85e04e9f46'
DISTRIB_TARGET='ramips/mt76x8'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.3 r11063-85e04e9f46'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.3"
VERSION_ID="19.07.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11063-85e04e9f46"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.3 r11063-85e04e9f46"

opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10

KB5201314, 2020-07-15:
看起来v2ray+upx在别的openwrt设备（MT7620A）上也遇到了问题：https://zyfdegh.github.io/post/202002-go-compile-for-mips/

这里写道：

> But the upx compressed one hang on command line, let’s skip this.
> ```
> root@OpenWrt:/tmp# ./v2ray --help
> <hang, nothing prints>
> ```

cyber386, 2020-08-01:
https://github.com/cyber386/v2ray-core-build-without-v2ctl 大家可以先到这里下载我编译的版本，直接替换二进制文件就行，都测试过。

closed 2020-08-02 by cyber386

kuoruan, 2020-08-02:
这是 UPX 的BUG，已经修复 https://github.com/upx/upx/issues/339#issuecomment-583903638
我会在下个版本做个更新