kuoruan/openwrt-v2ray #93: How to determine which encryption/security method does v2ray use when "security": "auto"

sgon00, 2020-07-02:
Hi, I am using vmess outbound protocol. Inside the `users` object, I have `"security": "auto",` setup.
I would like to know which encryption method the v2ray will use. 
I tried to change loglevel to `"debug"` and check the log file, But there is no such information in the log file.
How can I determine which encryption method the v2ray uses?
Thanks a lot.

kuoruan, 2020-07-02:
Please see V2Ray's document: https://www.v2ray.com/en/configuration/protocols/vmess.html#userobject

sgon00, 2020-07-02:
@kuoruan thanks a lot for the quick reply. I did see that document before asking this question. I knew that document/spec and what should happen. But what should happen does NOT mean what will happen. I mean the implementation may NOT follow the spec. I would like to see some log info or stdout to check if v2ray openwrt client does follow the doc's rule. Are there any log that will show the encryption method name? Thanks.

kuoruan, 2020-07-02:
Please head to V2Ray discussion to get some help: https://github.com/v2ray/discussion/issues

sgon00, 2020-07-02:
OK, thanks.

closed 2020-07-02 by sgon00