kuoruan/openwrt-v2ray #95: 不借助luci，v2ray-core-mini如何读取配置文件？

yijing1998, 2020-08-01:
Please attach the output of following commands.

```
cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.3'
DISTRIB_REVISION='r11063-85e04e9f46'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.3 r11063-85e04e9f46'
DISTRIB_TAINTS=''
root@OpenWrt-k2p:/etc# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10
```
因为空间有限，只安装了v2ray-core-mini。
安装v2ray-core-mini后，使用`v2ray -c /path/to/config.json`，终端没有输出任何信息，`ps`命令可看到进程存在。用`netstat`命令查看不到inbound的监听端口，感觉没有正常工作。

cyber386, 2020-08-01:
已知问题，一直没解决。https://github.com/kuoruan/openwrt-v2ray/issues/90
https://github.com/cyber386/v2ray-core-build-without-v2ctl 可以到这里下我编译的版本，直接替换掉就行了。

kuoruan, 2020-08-02:
UPX 的问题 https://github.com/upx/upx/issues/339#issuecomment-583903638

yijing1998, 2020-08-03:
能否用动态链接库编译？不太了解golang，瞎提些建议

closed 2020-08-03 by kuoruan