kuoruan/openwrt-v2ray #97: v2ray-core-mini found, but incompatible with the architectures configured - arm_cortex-a7

dsseter, 2020-08-17:
Please attach the output of following commands.

```
root@XiaoQiang:/tmp# opkg install v2ray-core-mini_4.27.0-1_arm_cortex-a7_neon-vfpv4.ipk
Unknown package 'v2ray-core-mini'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for v2ray-core-mini found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package v2ray-core-mini.
```

```
root@XiaoQiang:/tmp# cat /usr/lib/os-release
NAME="OpenWrt"
VERSION="18.06-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 18.06-SNAPSHOT"
VERSION_ID="18.06-snapshot"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="unknown"
LEDE_BOARD="ipq60xx/MiWiFi"
LEDE_ARCH="arm_cortex-a7"
LEDE_TAINTS="no-all busybox"
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt 18.06-SNAPSHOT unknown"
root@XiaoQiang:/tmp# cat /etc/openwrt_release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='18.06-SNAPSHOT'
DISTRIB_REVISION='unknown'
DISTRIB_TARGET='ipq60xx/MiWiFi'
DISTRIB_ARCH='arm_cortex-a7'
DISTRIB_DESCRIPTION='OpenWrt 18.06-SNAPSHOT unknown'
DISTRIB_TAINTS='no-all busybox'
root@XiaoQiang:/tmp#
```
现在有的 v2ray-core-mini_4.27.0-1_arm_cortex-a7_neon-vfpv4.ipk 在 DISTRIB_ARCH='arm_cortex-a7'
 架构下不能成功安装。
小米 AX5 路由器， 辛苦交叉编译一份， 万分感谢

suyanlong, 2020-11-01:
1

xiedeacc, 2021-10-03:
+1空间明明够

binary-person, 2024-10-31:
I have built a version for arm-cortex-a7 here:
https://github.com/binary-person/openwrt-v2ray/releases/download/v5.21.0-2/v2ray-core-mini_5.21.0-r1_arm_cortex-a7.ipk

But I get `__nanosleep_time64: symbol not found` on GL-AXT1800.

Using the following works instead
https://github.com/felix-fly/v2ray-openwrt/releases/download/v5.16.1/v2ray-linux-armv7.tar.gz