kuoruan/openwrt-v2ray #98: v2ray version >4.25 crashes on EdgeRouter-X

Tualua, 2020-08-26:
```
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/proxy/freedom/freedom.go:172 +0x144
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.OnSuccess.func1(0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:12 +0x44
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.Run.func1(0x35e0a40, 0x3dca0c0, 0x265b640)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:28 +0x3c
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: created by v2ray.com/core/common/task.Run
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:27 +0xf0
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]:
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: goroutine 15440 [select]:
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.Run(0xecdbe0, 0x22a5880, 0x3e587a8, 0x2, 0x2, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:42 +0x1e0
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/proxy/dokodemo.(*DokodemoDoor).Process.func5(0x3, 0xece540)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/proxy/dokodemo/dokodemo.go:206 +0x70
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.OnSuccess.func1(0x3, 0x1cc9bc0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:12 +0x44
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.Run.func1(0x32f8ae8, 0x3ea3c40, 0x36501c0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:28 +0x3c
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: created by v2ray.com/core/common/task.Run
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:27 +0xf0
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]:
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: goroutine 15441 [select]:
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/transport/pipe.(*pipe).ReadMultiBuffer(0x2363650, 0x6c31c4, 0x32f8b00, 0x0, 0x0, 0x32f8b00)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/transport/pipe/impl.go:89 +0xf4
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/transport/pipe.(*Reader).ReadMultiBuffer(0x32f8a88, 0x0, 0xcb9460, 0x1, 0x32f8b00, 0x6bcd90)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/transport/pipe/reader.go:16 +0x40
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.copyInternal(0xec2ec0, 0x32f8a88, 0xec1d80, 0x32f8ac8, 0x36501e0, 0xcb9460, 0x1)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/copy.go:81 +0x44
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.Copy(0xec2ec0, 0x32f8a88, 0xec1d80, 0x32f8ac8, 0x3da4fb4, 0x1, 0x1, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/copy.go:104 +0xd8
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/proxy/dokodemo.(*DokodemoDoor).Process.func4(0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/proxy/dokodemo/dokodemo.go:199 +0xe8
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.Run.func1(0x32f8ae8, 0x3ea3c40, 0x3ea3c00)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:28 +0x3c
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: created by v2ray.com/core/common/task.Run
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:27 +0xf0
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]:
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: goroutine 15537 [IO wait]:
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: internal/poll.runtime_pollWait(0x6fc1e9a8, 0x72, 0xffffffff)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	runtime/netpoll.go:203 +0x60
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: internal/poll.(*pollDesc).wait(0x3eb61a4, 0x72, 0x800, 0x800, 0xffffffff)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	internal/poll/fd_poll_runtime.go:87 +0x50
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: internal/poll.(*pollDesc).waitRead(...)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	internal/poll/fd_poll_runtime.go:92
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: internal/poll.(*FD).Read(0x3eb6190, 0x1938800, 0x800, 0x800, 0x0, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	internal/poll/fd_unix.go:169 +0x1ac
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: net.(*netFD).Read(0x3eb6190, 0x1938800, 0x800, 0x800, 0x1a001e0, 0x6628c8, 0x77e06750)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	net/fd_unix.go:202 +0x58
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: net.(*conn).Read(0x32f8ac0, 0x1938800, 0x800, 0x800, 0x0, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	net/net.go:184 +0x94
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.(*Buffer).ReadFrom(0x38f4f60, 0x6fb7d080, 0x32f8ac0, 0x77e06750, 0x0, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/buffer.go:175 +0x88
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.readOneUDP(0x6fb7d080, 0x32f8ac0, 0x1, 0x1, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/reader.go:13 +0xfc
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.(*PacketReader).ReadMultiBuffer(0x32f8ad0, 0x3a173c8, 0x1, 0x1, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/reader.go:169 +0x48
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.copyInternal(0xec1d40, 0x32f8ad0, 0xec2ee0, 0x32f8a70, 0x265b750, 0xcb9460, 0x1)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/copy.go:81 +0x44
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/buf.Copy(0xec1d40, 0x32f8ad0, 0xec2ee0, 0x32f8a70, 0x3dc3fb0, 0x1, 0x1, 0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/buf/copy.go:104 +0xd8
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/proxy/dokodemo.(*DokodemoDoor).Process.func3(0x0, 0x0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/proxy/dokodemo/dokodemo.go:188 +0x138
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: v2ray.com/core/common/task.Run.func1(0x35e0a80, 0x3dca240, 0x3ea3bc0)
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:28 +0x3c
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: created by v2ray.com/core/common/task.Run
Tue Aug 25 17:26:14 2020 daemon.err v2ray[25118]: 	v2ray.com/core/common/task/task.go:27 +0xf0
```

Please attach the output of following commands.

```
cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.3'
DISTRIB_REVISION='r11063-85e04e9f46'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07.3 r11063-85e04e9f46'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.3"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.3"
VERSION_ID="19.07.3"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11063-85e04e9f46"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.3 r11063-85e04e9f46"

opkg print-architecture

arch all 1
arch noarch 1
arch mipsel_24kc 10
```
Full log is attached
[system-v2ray.log](https://github.com/kuoruan/openwrt-v2ray/files/5127122/system-v2ray.log)

kuoruan, 2020-08-26:
可以去官方开个 issue 问问

phantomedc, 2020-08-31:
> 可以去官方开个 issue 问问

新版使用高版本Go编译，导致虚拟内存占用700M+，频繁崩溃，退回旧版本就只有100M+