kuoruan/openwrt-v2ray #99: 更新到4.28无法使用

naturecurly, 2020-09-12:
Please attach the output of following commands.

```
cat /etc/*release
opkg print-architecture
```

kuoruan, 2020-09-12:
有报错信息之类的吗

naturecurly, 2020-09-12:
> 有报错信息之类的吗

没有，我看服务器端很多人更新到28以后也不能运行了，所以我先退到27了，就好了。应该是覆盖更新的一些问题

nobk, 2020-09-13:
似乎是4.28读取配置文件后，对一些默认项目的取值和以前的版本不一样了，我这里三方v2r提供商的配置文件都用不了，我自己的vps的配置文件还可以用。退回4.27.1则全部正常。

alonewoo, 2020-09-13:
4.28 server端安装目录要先迁移，老方式服务器端看着像升级了，实际运行的还是老版本。参考官方升级指南。

naturecurly, 2020-09-13:
应该是luci-v2ray不兼容28版本了，命令行开启是可以的，但luci开启就一直显示未运行，是不是luci也要更新一下

alonewoo, 2020-09-13:
我还是在用luci-v2ray v1.5.6-1，openwrt 上的v2ray客户端也要4.28.0， alterId=0；我还没升4.28.1

oemaix, 2020-09-14:
我遇到一个升级到4.28以后无法启动的问题, 报错信息是
daemon.err v2ray[11033]: 2020/09/14 10:10:04 mmap: cannot allocate memory
4.27的时候 v2ray.main.mem_percentage 没设置就能跑.
升级4.28以后, 设置v2ray.main.mem_percentage='0' 就正常了.

closed 2020-10-04 by naturecurly