kuoruan/openwrt-v2ray #100: Package v2ray-core version 4.28-1 has no valid architecture

nielspeen, 2020-09-22:
After upgrading to OpenWRT 19.07.4 I'm no-longer able to install v2ray-core:

```
Package v2ray-core version 4.nn-n has no valid architecture, ignoring. Cannot install package v2ray-core.
```

This worked fine when I was using OpenWRT 19.07.2.

```
root@OpenWrt:~# cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07.4'
DISTRIB_REVISION='r11208-ce6496d796'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 19.07.4 r11208-ce6496d796'
DISTRIB_TAINTS=''
NAME="OpenWrt"
VERSION="19.07.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 19.07.4"
VERSION_ID="19.07.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r11208-ce6496d796"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 19.07.4 r11208-ce6496d796"

root@OpenWrt:~# opkg print-architecture
arch all 1
arch noarch 1
arch arm_cortex-a9_vfpv3-d16 10
```

kuoruan, 2020-09-22:
Which file did you use? `v2ray-core_4.28.1-1_arm_cortex-a9_vfpv3-d16.ipk` ?

nielspeen, 2020-09-22:
I did not try a specific package, but used the repository instead:

```
wget -O kuoruan-public.key http://openwrt.kuoruan.net/packages/public.key opkg-key add kuoruan-public.key
echo "src/gz kuoruan_packages http://openwrt.kuoruan.net/packages/releases/$(. /etc/openwrt_release ; echo $DISTRIB_ARCH)"   >> /etc/opkg/customfeeds.conf
echo "src/gz kuoruan_universal http://openwrt.kuoruan.net/packages/releases/all"   >> /etc/opkg/customfeeds.conf
opkg update
opkg install v2ray-core
```

kuoruan, 2020-09-22:
```
cat /etc/opkg/customfeeds.conf
```

check the path of feeds url.

nielspeen, 2020-09-22:
```
# add your custom package feeds here
#
# src/gz example_feed_name http://www.example.com/path/to/files
src/gz kuoruan_packages http://openwrt.kuoruan.net/packages/releases/arm_cortex-a9_vfpv3
src/gz kuoruan_universal http://openwrt.kuoruan.net/packages/releases/all
```

kuoruan, 2020-09-22:
OK, you are welcome.

closed 2020-09-22 by kuoruan